openbmc/openbmc #915: [Witherspoon - v1.99.0-353-ga3b8a0a] Driver 423: Intermittent Dbus exception during execution of Time related TC

sivassrr, 2017-01-05:
```
root@witherspoon:# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.0-353"
VERSION_ID="v1.99.0-353-ga3b8a0a-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.0-353"
BUILD_ID="v1.99.0"
root@witherspoon:# cat /etc/version
20161231134643
root@witherspoon:~# 
```

sivassrr, 2017-01-05:
[witherspoonDr423Iterations5TimeTCfails.pdf](https://github.com/openbmc/openbmc/files/686371/witherspoonDr423Iterations5TimeTCfails.pdf)

gkeishin, 2017-01-05:
I looked at the error and to me its more about timing when the SET and GET was issued. I see that within fraction of a second after setting the time mode, the GET was operation was also issued.. 

So if the SET operation takes time updating it to reflect then we should try adding a delay and try GET operation..

sivassrr, 2017-01-05:
[20170104163641080099_BMC_dmesg.txt](https://github.com/openbmc/openbmc/files/687393/20170104163641080099_BMC_dmesg.txt)
[20170104163641080099_BMC_general.txt](https://github.com/openbmc/openbmc/files/687391/20170104163641080099_BMC_general.txt)
[20170104163641080099_BMC_inventory.txt](https://github.com/openbmc/openbmc/files/687392/20170104163641080099_BMC_inventory.txt)
[20170104163641080099_BMC_journalctl.txt](https://github.com/openbmc/openbmc/files/687394/20170104163641080099_BMC_journalctl.txt)
[20170104163641080099_BMC_led.txt](https://github.com/openbmc/openbmc/files/687395/20170104163641080099_BMC_led.txt)
[20170104163641080099_BMC_meminfo.txt](https://github.com/openbmc/openbmc/files/687399/20170104163641080099_BMC_meminfo.txt)
[20170104163641080099_BMC_proc_list.txt](https://github.com/openbmc/openbmc/files/687397/20170104163641080099_BMC_proc_list.txt)
[20170104163641080099_BMC_procinfo.txt](https://github.com/openbmc/openbmc/files/687398/20170104163641080099_BMC_procinfo.txt)
[20170104163641080099_BMC_sensor_list.txt](https://github.com/openbmc/openbmc/files/687396/20170104163641080099_BMC_sensor_list.txt)

gkeishin, 2017-01-05:
@sivassrr Well I was just interpreting the log data from the test result perspective.. yes it may happen such scenario in other cases too . so the development folks should help explain here what is this error all about.

geissonator, 2017-01-05:
Here's what appears to be the relevant info from the journal for time testing:
```
Nov 21 19:23:55 witherspoon phosphor-gevent[2743]: 9.126.171.88 - - [2016-11-21 19:23:55] "POST /login HTTP/1.1" 200 320 0.030767
Nov 21 19:23:55 witherspoon phosphor-gevent[2743]: 9.126.171.88 - - [2016-11-21 19:23:55] "PUT /org/openbmc/settings/host0/attr/time_mode HTTP/1.1" 200 169 0.153298
Nov 21 19:23:57 witherspoon phosphor-gevent[2743]: 9.126.171.88 - - [2016-11-21 19:23:57] "POST /login HTTP/1.1" 200 320 0.031380
Nov 21 19:23:57 witherspoon phosphor-gevent[2743]: 9.126.171.88 - - [2016-11-21 19:23:57] "GET /org/openbmc/settings/host0/attr/time_mode HTTP/1.1" 200 170 0.064344
Nov 21 19:23:58 witherspoon phosphor-gevent[2743]: 9.126.171.88 - - [2016-11-21 19:23:58] "POST /login HTTP/1.1" 200 320 0.031057
Nov 21 19:23:59 witherspoon phosphor-gevent[2743]: 9.126.171.88 - - [2016-11-21 19:23:59] "PUT /org/openbmc/settings/host0/attr/time_mode HTTP/1.1" 200 169 0.146410
Nov 21 19:24:00 witherspoon phosphor-gevent[2743]: 9.126.171.88 - - [2016-11-21 19:24:00] "POST /login HTTP/1.1" 200 320 0.030830
Nov 21 19:24:00 witherspoon phosphor-gevent[2743]: 9.126.171.88 - - [2016-11-21 19:24:00] "PUT /org/openbmc/settings/host0/attr/time_owner HTTP/1.1" 200 169 0.146760
Nov 21 19:24:02 witherspoon phosphor-gevent[2743]: 9.126.171.88 - - [2016-11-21 19:24:02] "POST /login HTTP/1.1" 200 320 0.031108
Nov 21 19:24:02 witherspoon dbus[710]: [system] Activating via systemd: service name='org.freedesktop.timedate1' unit='dbus-org.freedesktop.timedate1.service'
Nov 21 19:24:02 witherspoon systemd[1]: Starting Time & Date Service...
Nov 21 19:24:02 witherspoon phosphor-gevent[2743]: 9.126.171.88 - - [2016-11-21 19:24:02] "PUT /org/openbmc/settings/host0/attr/time_mode HTTP/1.1" 200 169 0.199756
Nov 21 19:24:02 witherspoon dbus[710]: [system] Successfully activated service 'org.freedesktop.timedate1'
Nov 21 19:24:02 witherspoon systemd[1]: Started Time & Date Service.
Nov 21 19:24:02 witherspoon systemd-timedated[2784]: /etc/localtime should be a symbolic link to a time zone data file in /usr/share/zoneinfo/.
Nov 21 19:24:03 witherspoon systemd[1]: Reloading.
Nov 21 19:24:04 witherspoon phosphor-gevent[2743]: 9.126.171.88 - - [2016-11-21 19:24:04] "POST /login HTTP/1.1" 200 320 0.030904
Nov 21 19:24:04 witherspoon systemd[1]: Stopping Network Time Synchronization...
Nov 21 19:24:04 witherspoon systemd[1]: Stopped Network Time Synchronization.
Nov 21 19:24:04 witherspoon systemd-timedated[2784]: Set NTP to disabled
Nov 21 19:24:04 witherspoon systemd[1]: Stopped Network Time Synchronization.
Nov 21 19:24:04 witherspoon phosphor-gevent[2743]: 9.126.171.88 - - [2016-11-21 19:24:04] "GET /org/openbmc/settings/host0/attr/time_mode HTTP/1.1" 503 301 0.205174
```

sivassrr, 2017-01-12:
[WitherspoonDriver437Iterations10.pdf](https://github.com/openbmc/openbmc/files/701282/WitherspoonDriver437Iterations10.pdf)

``` All TCs for withers spoon except the TC - Set Manual Time Mode failed 7 out of 10 times. ```

sivassrr, 2017-01-12:
[20170112132120014605_BMC_dmesg.txt](https://github.com/openbmc/openbmc/files/701292/20170112132120014605_BMC_dmesg.txt)
[20170112132120014605_BMC_general.txt](https://github.com/openbmc/openbmc/files/701294/20170112132120014605_BMC_general.txt)
[20170112132120014605_BMC_inventory.txt](https://github.com/openbmc/openbmc/files/701293/20170112132120014605_BMC_inventory.txt)
[20170112132120014605_BMC_journalctl.txt](https://github.com/openbmc/openbmc/files/701295/20170112132120014605_BMC_journalctl.txt)
[20170112132120014605_BMC_led.txt](https://github.com/openbmc/openbmc/files/701299/20170112132120014605_BMC_led.txt)
[20170112132120014605_BMC_meminfo.txt](https://github.com/openbmc/openbmc/files/701296/20170112132120014605_BMC_meminfo.txt)
[20170112132120014605_BMC_proc_list.txt](https://github.com/openbmc/openbmc/files/701300/20170112132120014605_BMC_proc_list.txt)
[20170112132120014605_BMC_procinfo.txt](https://github.com/openbmc/openbmc/files/701298/20170112132120014605_BMC_procinfo.txt)
[20170112132120014605_BMC_sensor_list.txt](https://github.com/openbmc/openbmc/files/701297/20170112132120014605_BMC_sensor_list.txt)

sivassrr, 2017-01-13:
With driver 442 not seeing this error - Good news. Test 10 iterations not even a single failure. 
 Updated the report in https://github.com/openbmc/openbmc-test-automation/issues/220

gkeishin, 2017-05-26:
anything more on this to be done.. from my end I dont see this anymore.. if not can we close this ?

sivassrr, 2017-05-29:
Not seeing recently, So closing it.

closed 2017-05-29 by sivassrr