openbmc/openbmc #916: [Palmetto VERSION_ID="v1.99.0-360-g96d8e17"] Failed to boot the host

SrideviRamesh, 2017-01-05:
Issue:

BMC is not reaching to HOST_BOOTED after Updating BMC with latest image and PNOR with latest pnor

Log Info:

Logs collected form BMC journal logs

palmetto inventory_upload.py[986]: Uploading inventory to PNOR in dtb format...
palmetto inventory_upload.py[986]: Error 100 opening ffs !
palmetto inventory_upload.py[986]: FFS: Flash header not found. Code: 100 

system Info:

cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.0-360"
VERSION_ID="v1.99.0-360-g96d8e17-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.0-360"
BUILD_ID="v1.99.0"

please find journal logs attached
[journals.txt](https://github.com/openbmc/openbmc/files/686966/journals.txt)

amboar, 2017-01-08:
@SrideviRamesh Please use the triple  back-tick markdown when pasting code or console output into github issues. It prevents github's rendering engine interpreting your log output as markdown. What you write looks like:

    ```
    My code, console or log output here
    ```

amboar, 2017-01-08:
@SrideviRamesh Addressing the actual problem, are you sure the PNOR is not corrupt? Have you tried flashing a new image to it?

SrideviRamesh, 2017-01-09:
@amboar yes, i have used the latest PNOR image

SrideviRamesh, 2017-01-19:
Observing the same issue with palmetto build #v1.99.0-394 and latest pnor image...
please find the attached journal logs
[journal_palmetto.txt](https://github.com/openbmc/openbmc/files/716128/journal_palmetto.txt)

@amboar  please give updates

sivassrr, 2017-01-23:
This is duplicate of https://github.com/openbmc/openbmc/issues/861 
Requested to give high priority on 861/916. This will help us to get going on full CT.

closed 2017-01-23 by sivassrr