openbmc/openbmc #922: led: Support power supply LED's

geissonator, 2017-01-05:


sivassrr, 2017-04-28:
Is it going to come by phase5?

geissonator, 2017-11-28:
BMC will still  log error, and it will have appropriate callout, just not LED lit without this story.

spinler, 2017-11-29:
So far, all LEDs must show up in /sys/class/leds.  If that were to happen for this LED, the PS device driver would need to handle converting that into the I2C Commands to set/clear the LED.  Then, we'd just need to model that LED in the MRW with the appropriate LED attributes, and everything else would just work after that.

geissonator, 2018-01-15:
Per @spinler - Eddie has some driver changes coming to make the PS faule LED show up in /sys/class/leds, then we'll just need an MRW update.

rfrandse, 2018-02-16:
<https://gerrit.openbmc-project.xyz/9037> Witherspoon XML revision bump
Resolves  #922 led: Support power supply LED's

closed 2018-02-27 by geissonator

spinler, 2018-07-18:
For the sake of completeness, the MRW change I made here also adds in the I2C connections for the power supplies which means we'll get real inventory callouts on I2C fails now, instead of just device path callouts.