openbmc/openbmc #927: [Barreleye - v1.99.0-360-g96d8e17] - Present defect for membuf0 is missing

rahulmah, 2017-01-06:
In Berreleye, Present defect for membuf0 is missing.

Present defect for membuf0 with latest build
```
rahulmah@rackspace:~$ curl -c cjar -b cjar -khttps://xx.xx.xx.xx/org/openbmc/inventory/system/chassis/motherboard/membuf0
{
  "data": {
    "Custom Field 1": "",
    "Custom Field 2": "",
    "Custom Field 3": "",
    "Custom Field 4": "",
    "Custom Field 5": "",
    "Custom Field 6": "",
    "Custom Field 7": "",
    "Custom Field 8": "",
    "FRU File ID": "",
    "Manufacturer": "INGRASYS",
    "Name": "BARRELEYE",
    "Part Number": "1A4218800-600-G",
    "Serial Number": "64Q0004X02"       <----- Presence defect was present earlier
  },
  "message": "200 OK",
  "status": "ok"
```

Present defect for membuf0 with older build
```
curl -c cjar -b cjar -khttps://9.3.23.32/org/openbmc/inventory/system/chassis/motherboard/membuf0"
{
  "data": {
    "Custom Field 1": "ECID:019A007040086718080C000000D10471",
    "Custom Field 2": "",
    "Custom Field 3": "",
    "Custom Field 4": "",
    "Custom Field 5": "",
    "Custom Field 6": "",
    "Custom Field 7": "",
    "Custom Field 8": "",
    "FRU File ID": "",
    "Manufacturer": "0000000000000000",
    "Name": "0000000000000000",
    "Part Number": "0000000000000000",
    "Serial Number": "0000000000000000",
    "fault": "False",
    "fru_type": "MEMORY_BUFFER",
    "is_fru": 0,
    "present": "True",     <----- Presence defect was present earlier
    "version": ""
  },
  "message": "200 OK",
  "status": "ok"
```

BMC fw version: 
root@barreleye:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.0-360"
VERSION_ID="v1.99.0-360-g96d8e17-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.0-360"
BUILD_ID="v1.99.0"
root@barreleye:~# cat /etc/version
20170104214547
root@barreleye:~#

rahulmah, 2017-01-11:
We are continuously hitting this issue on one of our Barreleye system. Can we get this issue analysed for root cause?

gkeishin, 2017-05-26:
Inventory are now in xyz .. Closing this as no further needs to be done..

closed 2017-05-26 by gkeishin