openbmc/openbmc #930: host ipmi: soft power off with watchdog timers

geissonator, 2017-01-06:


vishwabmc, 2017-01-11:
@geissonator Please assign this to me

vishwabmc, 2017-01-11:
After looking into it, I don't vote for running a watchdog to ensure "SMS attention is processed" since there is no ack for that SMS_ATTN.

Post sending sms attention, probably just ONE watchdog timer could be started and that can be stopped as part of actual power off command and I need to workout the ways since we don't want to be hung waiting for watchdog timeout....

vishwabmc, 2017-01-11:
Given that its a priority, have we started seeing hot issues with this and hence want to fix?. To me, this looks like a nice to have.

geissonator, 2017-01-11:
The current power on/off code just does the hard shutdowns.  We want to switch it over to the soft shutdowns but we need to handle the case where the host is in a bad state, hence the reason this one is higher priority.

vishwabmc, 2017-01-12:
Task breakdown : 

[ ]

vishwabmc, 2017-02-08:
Should have updated this before.. So the idea is to write the own timer and then use it than reusing the watchdog timer.

closed 2017-03-30 by williamspatrick