openbmc/openbmc #940: Fan controller - Control functions

msbarth, 2017-01-09:
- [ ] Create fan control loop
- [ ] Add set fan speed based on inc/dec values
- [ ] Add queue up of speed inc requests from OCC and set delta fan speed on increase interval
- [ ] Add constant fan speed drop when no increase requested (on rpm drop interval)
- [ ] Add fan floor, including adjustment based on ambient(and/or other system states)
- [ ] Add fan ceiling, including adjustment based on ambient windows(and/or other system states)

bradbishop, 2017-04-24:
Initial swag guess at 5 pts for this.

rfrandse, 2017-06-13:
<https://gerrit.openbmc-project.xyz/4762> Allow generation of map types in events
Resolves  #940 Fan Control - Setting/adjusting fan floor

closed 2017-06-26 by williamspatrick