openbmc/openbmc #941: Assert fault LED's when association is set and cleared

geissonator, 2017-01-09:
The error logging component will eventually start associating errors with inventory items.  The LED code needs to monitor for this and assert the appropriate fault LED groups when this happens.  The association will also get cleared at some point, and the fault LED's need to be turned off.

#854 and #352 and #929 are all in this area.

vishwabmc, 2017-01-11:
This will be done after the callout section is implemented

vishwabmc, 2017-03-02:
@rfrandse Backend support ( the callout ) is still being developed and hence this should not be marked for current sprints

dhruvibm, 2017-03-23:
- [ ] Understand LED and inventory

- [ ] Assert LED when a callout is created

- [ ] DeAssert LED when callout is resolved

- [ ] Integration testing.

rfrandse, 2017-05-10:
https://gerrit.openbmc-project.xyz/3938 Adding fru fault monitor service
Resolves #941 Assert fault LED's when association is set and cleared

closed 2017-05-19 by williamspatrick