openbmc/openbmc #942: Disable automatic reboot on host checkstop detection

geissonator, 2017-01-09:
MFG and lab require the ability to disable the automated reboot that happens on a checkstop detection.  This feature will enable that function.  Need to define a property (host?) that can be set/cleared for this function.

sivassrr, 2017-01-16:
@geissonator This will be only applicable to MFG and lab setup, right. Customer would not see this.  Because Andrea was working/worked on reboot the host on detecting checkstop.

geissonator, 2017-01-20:
Per a technical discussion today, we're just going to make one generic setting that people can set which will disable any type of automatic reboot.  Re-titling and closing out #943

mtritz, 2017-01-26:
- [ ] Add new "Enable/disable automatic reboot" element to settingsd
- [ ] New app in host repo to enable or disable reboot based on that element - more details later
- [ ] Recipe/service file to get new app into the right place

closed 2017-02-23 by geissonator

spinler, 2017-02-23:
So how do I disable automatic reboots?

mtritz, 2017-02-23:
Set the auto_reboot settingsd parameter to "no" - as in
`busctl call org.openbmc.settings.Host /org/openbmc/settings/host0 org.freedesktop.DBus.Properties Set sss org.openbmc.settings.Host auto_reboot no`