openbmc/openbmc #949: [Barreley v1.99.0-360-g96d8e17] Unwanted characters in the BIOS Custom fields

gkeishin, 2017-01-09:
Example
             `"Custom Field 1": "\tbuildroot-51f1ac9", ` 

should be
               `"Custom Field 1": "buildroot-51f1ac9", `
and likewise other fields too..

rango@ubuntu:~/openbmc-test-automation$  curl -c cjar -b cjar -k -H "Content-Type: application/json" -X GET https://xx.xx.xx.xx/org/openbmc/inventorysystem/bios
{
  "data": {
    "Asset Tag": "", 
    `"Custom Field 1": "\tbuildroot-51f1ac9", `
    `"Custom Field 2": "\tskiboot-5.4.2", `
   ` "Custom Field 3": "\thostboot-e28b28f", `
    `"Custom Field 4": "\tlinux-4.4.39-openpower1-4ff880f", `
    `"Custom Field 5": "\tpetitboot-v1.4.0-c5f30a7", `
    `"Custom Field 6": "\tbarreleye-xml-253622f", `
    `"Custom Field 7": "\tocc-d7efe30", `
    `"Custom Field 8": "\thostboot-binaries-56532f5", `
    "FRU File ID": "", 
    "Manufacturer": "", 
    "Model Number": "", 
    "Name": "OpenPOWER Firmware", 
    "Serial Number": "", 
    "Version": "open-power-barreleye-3c69277", 
    "fault": "False", 
    "fru_type": "SYSTEM", 
    "is_fru": 1, 
    "present": "True", 
    "version": ""
  }, 
  "message": "200 OK", 
  "status": "ok"

williamspatrick, 2017-01-10:
Is this a new problem or has it always been this way?

gkeishin, 2017-01-11:
Yeah,, I have seen this earlier also.. so this seems to has been always like this with this extra char "\t"

gkeishin, 2017-03-10:
This is no longer valid since the new software manager versioning is the equivalent of this in xyz.. Closing

closed 2017-03-10 by gkeishin