openbmc/openbmc #950: [Barreley v1.99.0-361-g8ce17d1] State Manager Core dumped

gkeishin, 2017-01-10:
Currently we are patching the state manager changes onto the new driver and testing it.

We were doing basic test power on and off 

``` [2]+  Aborted                 (core dumped) ./op-host-state-manager  (wd: /tmp)```

`root@barreleye:~# ps -aef | grep op-| grep -v grep`
 `925 root       0:00 ./op-chassis-state-manager`
`root@barreleye:~#`

So the host crashed,..

- [x] Journal log attached  
[journald_v1.99.0-361-g8ce17d1.txt](https://github.com/openbmc/openbmc/files/696229/journald_v1.99.0-361-g8ce17d1.txt)

geissonator, 2017-01-10:
I hit this as well once in testing, I saw this to the console when it happened:

[barreleye]# Assertion 'm->n_ref > 0' failed at ../git/src/libsystemd/sd-bus/bus-message.c:936, function sd_bus_message_unref(). Aborting.

My current theory is that maybe it's something in my code that handles the system state signals. 

https://github.com/openbmc/phosphor-state-manager/blob/master/host_state_manager.cpp#L91

williamspatrick, 2017-01-10:
```
int Host::handleSysStateChange(sd_bus_message *msg, void *usrData,
                               sd_bus_error *retError)
{
    const char *newState = nullptr;
    auto sdPlusMsg = sdbusplus::message::message(msg);  <-----
    sdPlusMsg.read(newState);
```

This line should be `... sdbusplus::message::message(sd_bus_message_ref(msg));`

geissonator, 2017-01-11:
tps://gerrit.openbmc-project.xyz/#/c/1749/

geissonator, 2017-01-13:
New proposed fix in sdbusplus from Patrick here - https://gerrit.openbmc-project.xyz/#/c/1778/

closed 2017-01-13 by williamspatrick