openbmc/openbmc #952: Need support for I2C Block Writes and Reads

tsawan1, 2017-01-11:
We have card drivers that require I2C Block read capability (up to 256 bytes) but it seems that feature is lacking in i2cget but does exist in i2cset in openbmc I2C toolset.

It seems that i2cset has that capability and am hoping someone in this community with better skills then me could adopt it in i2cget. Furthermore, I have located this patch on git that implements block read for i2cget, and am hoping it could be helpful in this endeavour.

Here is a sample command of what we expect for the updated i2cget with block capability:
-- Access I2C bus 2 and Read 64 bytes consecutively from address 0x77 with address offset  of 0:
i2cget -y 2 0x77 0 -r 0-63 i;

-- For reference, here is a real example of Access to I2C bus 2 and Write a block of 8 bytes (i option) starting at address 0x77 with address offset  of 0: 
i2cset -r -y 2 0x77 0 0x10 0x41 0x00 0x40 0x1B 0x50 0xFF 0xD5 i

vishwabmc, 2017-01-12:
@eddiejames Is that something you could help with ?.

@spinler I searched openbmc repo and I see some of the scripts do use i2cget / i2cset. Do you have any pointers on this specific request ?

spinler, 2017-01-12:
These tools come from Yocto:  ./import-layers/yocto-poky/meta/recipes-devtools/i2c-tools/i2c-tools_3.1.2.bb

A few months ago someone made another fix to i2cget via a patch, so I guess we'd go the same route for this.
That was done with:
./meta-phosphor/common/recipes-devtools/i2c-tools/i2c-tools_%.bbappend
./meta-phosphor/common/recipes-devtools/i2c-tools/i2c-tools/0001-4-byte-read-support-466.patch

williamspatrick, 2017-01-12:
Is this for an eeprom or something else?  EPROMs already have support for larger transfers via the native drivers. What is the use case for large transfers otherwise and which devices do we need support for it with?

tsawan1, 2017-01-15:
We need it for one of our data acquisition test cards deployed manufacturing and lab for HW validation. The microcode requires block reads to enable real time data sampling.

williamspatrick, 2017-01-24:
@geissonator - We need to get this scheduled.

@tsawan1 - Do you have a timeline you need this for?

tsawan1, 2017-01-24:
We need it by February 6 to enable Module System Test mid February

spinler, 2017-01-25:
An earlier patch to do this is at https://patchwork.ozlabs.org/patch/622104/, but it looks like it never got anywhere.  I'll see how that works.

spinler, 2017-01-26:
https://gerrit.openbmc-project.xyz/#/c/2036/

closed 2017-01-31 by williamspatrick