openbmc/openbmc #954: [Barreleye v1.99.0-361-g8ce17d1] BMC state behaviour

gkeishin, 2017-01-11:
Two different behavior noticed while testing with the new State Manager patches on Barreleye

``` 
1. Poweron the Host to Running state,   login to Host OS and execute reboot
--> BMC ReIPL back to running and HOST booted

2. Poweron the Host to Running state , login to Host OS and "sudo shutdown -f"
-->  Host OS shutdown.. and BMC powered off

BMC Power off implies Chassis state to Off + Host state to Off
```
This is just a place holder to understand what we are seeing in 1 and 2 responses behavior are correct or not as per design..

As per discussion, BMC should have never power off in these cases. As per Chris input, a watchdog could have hard reset it or a software app could reset it, but it should always be keep running all the time.

- [x] Journal Log attached  
[BMC_powerstate_journald_v1.99.0-361-g8ce17d1.txt](https://github.com/openbmc/openbmc/files/698323/BMC_powerstate_journald_v1.99.0-361-g8ce17d1.txt)

sivassrr, 2017-01-11:
``` 
Actually BMC will not power off, but CEC / HOST should not lose the power / re-ipl in the case 1. 
BMC power should be in power on state. Hope we are not observing BMC -reset. 
And expectation of BMC state should be in READY (according to the new state defn. ). 
```

vishwabmc, 2017-01-11:
@gkeishin IMO, this looks right and I probably would not term it `2 different behaviours` as that would mean 2 results for `one testcase` but then its not the case here.

@geissonator I am still puzzled on the chassis power on / off... With host off, what of chassis do we want to differently handle ?.

geissonator, 2017-03-17:
Currently we only support hard power off and hard reboot.  There is a story coming, #1310 which will offer the warm and soft aspects which I think is what you're asking here @gkeishin ?

gkeishin, 2017-03-17:
Yes that's what we are looking for. @geissonator

geissonator, 2017-03-22:
ok, closing this, work tracked under #1310

closed 2017-03-22 by geissonator