openbmc/openbmc #957: Cache host inventory in filesystem

bradbishop, 2017-01-11:


dkodihal, 2017-07-27:
- [ ] Update pimgen.py to generate code to do the load()/save() that Cereal required. This needs to be generated for every inventory interface that we're interested in. The load()/save() can work with the underlying sdbusplus type.
- [ ] When PIM Notify() creates inventory objects, serialize them to a persistent location by calling save().
- [ ] When PIM starts, deserialize persisted objects and do a Notify()? How will we know which interfaces to apply on the object? Store the input to Notify()?

dkodihal, 2017-07-28:
# **STATUS**

rfrandse, 2017-08-09:
<https://gerrit.openbmc-project.xyz/5877> pimgen: ignore xyz.openbmc_project.Assocation
Resolves  #957 Cache inventory in filesystem

closed 2017-08-25 by williamspatrick