openbmc/openbmc #959: Fan control - Presence detection

msbarth, 2017-01-11:
- [ ] Check for fan presence(by tach feedback = 0) at poweron
- [ ] Add fan monitoring loop
- [ ] Check for fan presence(by tach feedback = 0) within monitoring loop
- [ ] Log error on fan detected missing
- [ ] Shutdown system with 1 or more fans missing

msbarth, 2017-01-11:
- [ ] Create fan control/monitoring repository
- [ ] Get fan monitoring application framework in place
- [ ] Determine number of fans and locations within the system
- [ ] Check for fan presence(by tach feedback = 0) at poweron
- [ ] Log error for any fans missing and shutdown system(assuming this doesnt break lab systems)
- [ ] Create recipe to include into BMC image and start at power on

geissonator, 2017-01-12:
Sprint planning notes
- Use Brad's new dbus objects to read fans
- Need meeting tomorrow with Brad/Patrick/Matt to discuss.

msbarth, 2017-01-16:
Determine, for the time being, how to shutdown the system due to any fans missing when inventory objects created when no tach is found after task list items complete.

closed 2017-03-15 by williamspatrick