openbmc/openbmc #960: [palmetto VERSION_ID="v1.99.0-377"] Presence Detect failing on things that exist system, chassis

SrideviRamesh, 2017-01-12:
Expected Behavior:
Presence Detect failing on things that exist system, chassis

Actual Behavior:
presence detect should be True on things that exist system, chassis

Log Info:
```
-bash-4.1$ curl -c cjar -b cjar -k https://9.3.23.30//org/openbmc/inventory/system
{
  "data": {
    "fault": "False",
    "fru_type": "SYSTEM",
    "is_fru": 1,
    "present": "True",
    "version": ""
  },
  "message": "200 OK",
  "status": "ok"
}

-bash-4.1$ curl -c cjar -b cjar -k https://9.3.164.177//org/openbmc/inventory/system
{
  "data": {
    "Asset Tag": "",
    "Custom Field 1": "\tbuildroot-211bd05",
    "Custom Field 2": "\tskiboot-5.3.6",
    "Custom Field 3": "\thostboot-3ed4df7",
    "Custom Field 4": "\tlinux-4.4.21-openpower1-d43e553",
    "Custom Field 5": "\tpetitboot-v1.3.0-d1ee745",
    "Custom Field 6": "\tpalmetto-xml-c6f5639-d76d973",
    "Custom Field 7": "\tocc-bb092e9",
    "Custom Field 8": "\thostboot-binaries-7f593a3",
    "FRU File ID": "",
    "Manufacturer": "",
    "Model Number": "",
    "Name": "OpenPOWER Firmware",
    "Serial Number": "",
    "Version": "open-power-palmetto-922a1c1",
    "fault": "False",
    "fru_type": "SYSTEM",
    "is_fru": 1,
    "present": "False",
    "version": ""
  },
  "message": "200 OK",
  "status": "ok"
}
```

Specifications:

cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.0-377"
VERSION_ID="v1.99.0-377-ge4c152a-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.0-377"
BUILD_ID="v1.99.0"

FFDC & journal logs attached
[20170112030018170484_SystemPresent.zip](https://github.com/openbmc/openbmc/files/701165/20170112030018170484_SystemPresent.zip)
[20170112025823681253_ChassisPresent.zip](https://github.com/openbmc/openbmc/files/701168/20170112025823681253_ChassisPresent.zip)
[test_chassis_journal.txt](https://github.com/openbmc/openbmc/files/701170/test_chassis_journal.txt)
[test_sensor_journal.txt](https://github.com/openbmc/openbmc/files/701172/test_sensor_journal.txt)

vishwabmc, 2017-01-12:
@SrideviRamesh Have you tried this on the same build level ?

SrideviRamesh, 2017-01-12:
@vishwabmc i have tried on the build level mentioned in the log

vishwabmc, 2017-01-12:
@SrideviRamesh In your first post, it just had : 

-bash-4.1$ curl -c cjar -b cjar -k https://xx.xx.xx.xx//org/openbmc/inventory/system
{
  "data": {
    "fault": "False",
    "fru_type": "SYSTEM",
    "is_fru": 1,
    "present": "True",
    "version": ""
  },
  "message": "200 OK",
  "status": "ok"
}

and what I see now is different. Which is the right one ?

vishwabmc, 2017-01-12:
@williamspatrick Looking at the original issue : openbmc/openbmc#93, you have this quote there

`Palmetto doesn't have a VPD chip on the backplane that the BMC can access. That is why it is missing the call you mentioned.`

So should we just ignore this testcase for good on Palmetto ?. @SrideviRamesh told me that she ran this testcase today and saw the failure.  Its an independent testcase and does not cause any issues to rest of their tests

gkeishin, 2017-05-26:
Inventory are now in xyz. no further action needed on this..

closed 2017-05-26 by gkeishin