openbmc/openbmc #963: Failure to build pkgconfig-native with gcc-6.2 on master

amboar, 2017-01-13:
`master` at commit 4a2201ff2890167b8d770fcd80783bff72a85382 fails to build `pkgconfig-native` with recent GCCs due to the presence of `-Werror=format-nonliteral` and `-Werror=unused-variable`

The bug is partly captured on [f.d.o's bugzilla](https://bugs.freedesktop.org/show_bug.cgi?id=95326)

Patches are provided in the linked bug as a work-around.

amboar, 2017-01-13:
As a nasty hack I did this:

```patch
diff --git a/import-layers/yocto-poky/meta/recipes-devtools/pkgconfig/pkgconfig_git.bb b/import-layers/yocto-poky/meta/recipes-devtools/pkgconfig/pkgconfig_git.bb
index d34f608f119d..d0e3d8c4d488 100644
--- a/import-layers/yocto-poky/meta/recipes-devtools/pkgconfig/pkgconfig_git.bb
+++ b/import-layers/yocto-poky/meta/recipes-devtools/pkgconfig/pkgconfig_git.bb
@@ -28,6 +28,9 @@ EXTRA_OECONF = "--without-internal-glib"
 EXTRA_OECONF_class-native = "--with-internal-glib"
 EXTRA_OECONF_class-nativesdk = "--with-internal-glib"
 
+CFLAGS_class-native += " -Wno-error=format-nonliteral -Wno-error=unused-variable"
+CFLAGS_class-nativesdk += " -Wno-error=format-nonliteral -Wno-error=unused-variable"
+
 acpaths = "-I ."
 
 BBCLASSEXTEND = "native nativesdk"
```

amboar, 2017-01-13:
See #702

PavamanSubramaniyam, 2017-03-20:
I am also facing this issue on a Ubuntu 17.04 LE OS installed on a P8 Hardware.

| ppc64le-linux-libtool: compile:  gcc -DHAVE_CONFIG_H -I. -I../../../git/glib/glib -I.. -I.. -I../glib -I../../../git/glib/glib -I../../../git/glib -DG_LOG_DOMAIN=\"GLib\" -DG_DISABLE_CAST_CHECKS -DGLIB_COMPILATION -DPCRE_STATIC -isystem/home/ubuntu/openbmc/build/tmp/sysroots/ppc64le-linux/usr/include -pthread -Wall -Wstrict-prototypes -Werror=declaration-after-statement -Werror=missing-prototypes -Werror=implicit-function-declaration -Werror=pointer-arith -Werror=init-self -Werror=format-security -Werror=format=2 -Werror=missing-include-dirs -fvisibility=hidden -isystem/home/ubuntu/openbmc/build/tmp/sysroots/ppc64le-linux/usr/include -O2 -pipe -c ../../../git/glib/glib/glib-unix.c -o libglib_2_0_la-glib-unix.o
| ../../../git/glib/glib/gdate.c: In function 'g_date_strftime':
| ../../../git/glib/glib/gdate.c:2497:7: error: format not a string literal, format string not checked [-Werror=format-nonliteral]
|        tmplen = strftime (tmpbuf, tmpbufsize, locale_format, &tm);
|        ^~~~~~
| ../../../git/glib/glib/gbacktrace.c: In function 'g_on_error_query':
| ../../../git/glib/glib/gbacktrace.c:177:5: warning: ignoring return value of 'fgets', declared with attribute warn_unused_result [-Wunused-result]
|      fgets (buf, 8, stdin);
|      ^~~~~~~~~~~~~~~~~~~~~
| ../../../git/glib/glib/gbacktrace.c: In function 'stack_trace':
| ../../../git/glib/glib/gbacktrace.c:301:18: warning: ignoring return value of 'dup', declared with attribute warn_unused_result [-Wunused-result]
|        close (0); dup (in_fd[0]);   /* set the stdin to the in pipe */
|                   ^~~~~~~~~~~~~~
| ../../../git/glib/glib/gbacktrace.c:302:18: warning: ignoring return value of 'dup', declared with attribute warn_unused_result [-Wunused-result]
|        close (1); dup (out_fd[1]);  /* set the stdout to the out pipe */
|                   ^~~~~~~~~~~~~~~
| ../../../git/glib/glib/gbacktrace.c:303:18: warning: ignoring return value of 'dup', declared with attribute warn_unused_result [-Wunused-result]
|        close (2); dup (out_fd[1]);  /* set the stderr to the out pipe */
|                   ^~~~~~~~~~~~~~~
| ../../../git/glib/glib/gbacktrace.c:318:3: warning: ignoring return value of 'write', declared with attribute warn_unused_result [-Wunused-result]
|    write (in_fd[1], "backtrace\n", 10);
|    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
| ../../../git/glib/glib/gbacktrace.c:319:3: warning: ignoring return value of 'write', declared with attribute warn_unused_result [-Wunused-result]
|    write (in_fd[1], "p x = 0\n", 8);
|    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
| ../../../git/glib/glib/gbacktrace.c:320:3: warning: ignoring return value of 'write', declared with attribute warn_unused_result [-Wunused-result]
|    write (in_fd[1], "quit\n", 5);
|    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
| In file included from ../../../git/glib/glib/ggettext.c:30:0:
| ../../../git/glib/glib/ggettext.c: In function 'ensure_gettext_initialized':
| ../../../git/glib/glib/glibintl.h:37:42: warning: statement with no effect [-Wunused-value]
|  #define bindtextdomain(Domain,Directory) (Domain)
|                                           ^
| ../../../git/glib/glib/ggettext.c:109:7: note: in expansion of macro 'bindtextdomain'
|        bindtextdomain (GETTEXT_PACKAGE, GLIB_LOCALE_DIR);
|        ^~~~~~~~~~~~~~
| cc1: some warnings being treated as errors
| Makefile:1392: recipe for target 'libglib_2_0_la-gdate.lo' failed
| make[6]: *** [libglib_2_0_la-gdate.lo] Error 1
| make[6]: *** Waiting for unfinished jobs....
| make[6]: Leaving directory '/home/ubuntu/openbmc/build/tmp/work/ppc64le-linux/pkgconfig-native/0.29+gitAUTOINC+5914edfe96-r0/build/glib/glib'
| Makefile:1935: recipe for target 'all-recursive' failed
| make[5]: *** [all-recursive] Error 1
| make[5]: Leaving directory '/home/ubuntu/openbmc/build/tmp/work/ppc64le-linux/pkgconfig-native/0.29+gitAUTOINC+5914edfe96-r0/build/glib/glib'
| Makefile:954: recipe for target 'all' failed
| make[4]: *** [all] Error 2
| make[4]: Leaving directory '/home/ubuntu/openbmc/build/tmp/work/ppc64le-linux/pkgconfig-native/0.29+gitAUTOINC+5914edfe96-r0/build/glib/glib'
| Makefile:1047: recipe for target 'all-recursive' failed
| make[3]: *** [all-recursive] Error 1
| make[3]: Leaving directory '/home/ubuntu/openbmc/build/tmp/work/ppc64le-linux/pkgconfig-native/0.29+gitAUTOINC+5914edfe96-r0/build/glib'
| Makefile:771: recipe for target 'all' failed
| make[2]: *** [all] Error 2
| make[2]: Leaving directory '/home/ubuntu/openbmc/build/tmp/work/ppc64le-linux/pkgconfig-native/0.29+gitAUTOINC+5914edfe96-r0/build/glib'
| Makefile:698: recipe for target 'all-recursive' failed
| make[1]: *** [all-recursive] Error 1
| make[1]: Leaving directory '/home/ubuntu/openbmc/build/tmp/work/ppc64le-linux/pkgconfig-native/0.29+gitAUTOINC+5914edfe96-r0/build'
| Makefile:457: recipe for target 'all' failed
| make: *** [all] Error 2
| ERROR: oe_runmake failed
| ERROR: Function failed: do_compile (log file is located at /home/ubuntu/openbmc/build/tmp/work/ppc64le-linux/pkgconfig-native/0.29+gitAUTOINC+5914edfe96-r0/temp/log.do_compile.12499)
ERROR: Task 931 (virtual:native:/home/ubuntu/openbmc/meta/recipes-devtools/pkgconfig/pkgconfig_git.bb, do_compile) failed with exit code '1'

PavamanSubramaniyam, 2017-03-20:
The fixes and patches which are pushed here aren't merged yet to the sources.

williamspatrick, 2017-03-28:
This will be resolved by the upgrade to Yocto 2.2 which has support for this newer GCC.  That work is nearly complete. In the meantime you can use a distro in the support list for Yocto 2.1, which is what we are based on top of, or you can use the Docker scripts we provide for CI testing to create a container for development.

amboar, 2018-05-23:
We're at Yocto 2.4 now and i haven't seen any such issues. Closing

closed 2018-05-23 by amboar