openbmc/openbmc #964: Romulus RTC driver not working

mine260309, 2017-01-13:
Romulus' RTC driver is not working well:

```
### journalctl -f&
hwclock --show
hwclock: RTC_RD_TIME: Invalid argument
Nov 29 06:17:40 romulus kernel: rtc-rv8803 11-0032: Voltage low, data is invalid.
Nov 29 06:17:40 romulus kernel[644]: rtc-rv8803 11-0032: Voltage low, data is invalid.
```

It prevent time related services running correctly, e.g. timedatectl fails with:
```
timedatectl
Nov 29 06:19:03 romulus dbus[646]: [system] Activating via systemd: service name='org.freedesktop.timedate1' unit='dbus-org.freedesktop.timedate1.service'
Nov 29 06:19:03 romulus systemd[1]: Starting Time & Date Service...
Nov 29 06:19:03 romulus dbus[646]: [system] Successfully activated service 'org.freedesktop.timedate1'
Nov 29 06:19:03 romulus systemd-timedated[787]: /etc/localtime should be a symbolic link to a time zone data file in /usr/share/zoneinfo/.
Nov 29 06:19:03 romulus systemd[1]: Started Time & Date Service.
Failed to query server: Invalid argument
Nov 29 06:19:03 romulus kernel: rtc-rv8803 11-0032: Voltage low, data is invalid.
Nov 29 06:19:03 romulus kernel[644]: rtc-rv8803 11-0032: Voltage low, data is invalid.
```

deece, 2017-01-13:
Is this running in an emulator?

shenki, 2017-01-13:
No, this is on Romulus

deece, 2017-01-13:
The datasheet says that the V2F (low voltage bit) is set at(RTC) power on and must be manually cleared.

The driver will clear this bit when the time is set, so cloud you please try manually setting the RTC time on the machine and see if the problem persists?

mine260309, 2017-01-13:
Yup, `hwclock --systohc` fixed the issue.
Now hwclock and timedatectl both work fine.

mine260309, 2017-01-13:
@williamspatrick please close this issue.

vishwabmc, 2017-01-13:
@mine260309 IMO, we still need this open.

@shenki Wonder why is this not an issue with Barreleye and others ?. Is it a different RTC used here in Romulus ?. If this is unique to Romulus, then we need some ways of clearing it automatically when we boot. Any suggestions on how this was done before ?

shenki, 2017-01-13:
Yes, Witherspoon and Romulus have a different RTC.

@mine260309 what was the root cause of the issue?

deece, 2017-01-13:
The low voltage bit will only be set if the RTC has been powered down (not the system, assuming the RTC has a battery). It does not need to be cleared at boot, it's basically saying that any data stored in the RTC is garbage.

vishwabmc, 2017-01-13:
@deece So if RTC looses battery power, then it looses its value.. Fine.. but once that is put on or is powered by AC, then there should be some means of correcting it instead of someone having to manually set it. Like via some init scripts etc ?

deece, 2017-01-13:
It's cleared when the RTC gets new valid data (by setting the time on it). If you clear it automatically, the user has no idea that anything has gone wrong (until some time down the track when they realise the timestamps are all offset).

vishwabmc, 2017-01-13:
@mine260309 I thought you were not able to set the time too ?

mine260309, 2017-01-13:
@vishwabmc After the time is set to RTC (`hwclock --systohc`), everything related to time is working fine.
So I expect this issue can be closed.

The root cause of the issue I guess is that the RTC on the Romulus board is **never** set with a valid time, so it has the low voltage bit. After it's cleared everything works fine.

If we want to reproduce the issue, the battery of the RTC needs to be removed.

vishwabmc, 2017-01-13:
@mine260309 This worked because that command was executed by hand but what I am recommending is that we need a way to auto set it since we can not be executing it on all.

mine260309, 2017-01-13:
@vishwabmc Although I have not tried yet, I believe this should be also done by systemd's timedatectl with `set-local-rtc` option, or with the `timedate1` service.
After time-manager is implemented, setting BMC time should make it sync to RTC, then the issue will not reproduce.
That's why I think this issue can be closed.

closed 2017-01-13 by williamspatrick

madscientist159, 2018-08-29:
We're seeing this same issue on Talos II (derived from Romulus).  Setting the date works, but cutting power to the system (leaving the battery installed) results in low voltage warnings etc.  Confirmed a solid 3V from the battery at the RTC battery pin, so I highly doubt this is a hardware problem.

```
[    3.993456] rtc-rv8803 11-0032: Voltage low, temperature compensation stopped.
[    4.000824] rtc-rv8803 11-0032: Voltage low, data loss detected.
[    4.007812] rtc-rv8803 11-0032: Voltage low, data is invalid.
[    4.014750] rtc-rv8803 11-0032: registered as rtc0
[    8.140247] rtc-rv8803 11-0032: Voltage low, data is invalid.
[    8.146030] rtc-rv8803 11-0032: hctosys: unable to read the hardware clock
```

mine260309, 2018-08-29:
@madscientist159 Does it occur for every time when it boots from AC cycle?
If yes, I doubt if the RTC is ever synced, it gets synced when:
1. If BMC's time setting is configured as NTP, and BCM successfully synced time from NTP server;
2. If you manually run `hwclock --systohc`

I saw such issue on brand a new system, when the RTC is not synced at all, it gets the above error from AC cycle, and is fixed by sync the RTC.

madscientist159, 2018-08-29:
@mine260309 That's what was weird about it -- I ran a manual sync, then pulled AC power.  On reattaching AC power, the data was not retained.

madscientist159, 2018-11-18:
Just as a follow up to this, it turns out ```hwclock --systohc``` has to be run at least once on the BMC.  Running it from the host has no effect whatsoever, even though the host can read the hardware clock directly via ```hwclock```.