openbmc/openbmc #965: IPL: Need OpenBMC command-line method to set 'bootdev none'

zmoose, 2017-01-13:
After loading firmware on a Barreleye (920.1702.20170110a) and booting the system to Petitboot the autoboot host OS doesn't happen.  Talking to Jeremy Kerr this is due to the following setting being enabled:
     Temporary IPMI boot option: Optical

We need a command-line method to clear this IPMI boot optical option so Petitboot's autoboot will automatically boot the host OS hard disk.  This would be the OpenBMC equivilant of 'ipmitool -I lanplus -H <BMC_hostname> -U ADMIN -P admin chassis bootdev none options=persistent' on OpenPower. 

Below is my sametime conversation with Jeremy Kerr on this topic.

Jan 12, 2017

dlshaw: 
6:32:16 PM: Hey Jeremy. Last night you mentioned in slack:

@dlshaw: if autoboot isn't enabled in petitboot, it won't autoboot at all (this isn't overrideable via IPMI)

[10:53]  
I'm not sure what you mean by "or if it's booting to "Optical"" though - the `bootdev none` will clear that setting.

The "Optical" thing I mentioned was "Persistent IPMI boot option: Optical".. Below is a screen copy of petitboot showing that option.

Petitboot System Configuration
 ??????????????????????????????????????????????????????????????????????????????

  Autoboot:      ( ) Disabled
                  Enabled

  Boot Order     (0) disk: sda2 [uuid: e250b90b-d2fa-42e8-be46-66f352ba041d]    

                 [    Add Device:     ]
                 [  Clear & Boot Any  ]
                 [       Clear        ]

  Timeout:       10    seconds

  Persistent IPMI boot option: Optical
  Clear option:  [ ]

  Network:       (*)CP on all active interfaces
                 ( ) DHCP on a specific interface
                 ( ) Static IP configuration

 ??????????????????????????????????????????????????????????????????????????????
 tab=next, shift+tab=previous, x=exit, h=help 

 

Jeremy Kerr (petitboot): 
6:32:45 PM: right, that's the current IPMI setting  

dlshaw: 
6:32:47 PM: Basically we have a BarelEye system that when code is flashed on it doesn't auto boot and trying to figure out why 

Jeremy Kerr (petitboot): 
6:33:01 PM: (ie, someone has done a 'chassis bootdev optical' previously)  

dlshaw: 
6:33:09 PM: Where does one run that command?  

6:33:12 PM: on OpenBMC 

6:33:31 PM: I tried running:
obmcutil chassis bootdev none options=persistent

But got failure msg 

Jeremy Kerr (petitboot): 
6:33:47 PM: I'm not entirely sure - if there's not yet an IPMI interface, it'll be somewhere in the REST interface  

6:34:03 PM: (the OpenBMC management stack is another team)  

6:35:09 PM: ('chassis bootdev xxx' is pretty specific to IPMI - I'm assuming someone would have implemented a REST equivalent)  

dlshaw: 
6:35:19 PM: Ya I hope so 

6:35:35 PM: is there a slack channel to ask that question in? 

Jeremy Kerr (petitboot): 
6:35:51 PM: but that'd definitely be why it isn't booting - the (internal) IPMI query that petitboot is doing is telling petitboot to only boot from optical devices.  

6:35:56 PM: just #openbmc  

dlshaw: 
6:36:04 PM: ok 

6:36:10 PM: Thanks will ask there  

Jeremy Kerr (petitboot): 
6:36:30 PM: no problem. let me know if you need me to do some digging myself.  

dlshaw: 
6:36:41 PM: Thanks will do 

6:40:35 PM: Seems like even when I do the "Clear option" 
 Temporary IPMI boot option: Optical
  Clear option:  [ ]

and then save it and check again its gone but when we reboot the system it stops at petiboot again and the "Temp IPMI boot option: optical" is back...

Crazy! 

6:40:53 PM: Its like I can't manually configure the system to just always autoboot the OS when we reboot the system 

Jeremy Kerr (petitboot): 
6:41:33 PM: the IPMI options always take precedence, to avoid unbootable systems (ie, the boot behaviour is always remote-controllable)  

dlshaw: 
6:41:58 PM: So I'll have to find the REST or other command to disable that option for us to autoboot the host OS? 

Jeremy Kerr (petitboot): 
6:42:05 PM: so if the BMC is telling firmware to boot from optical, that's not overridden by internal settings.  

6:42:06 PM: yep  

dlshaw: 
6:42:08 PM: Gotcha 

6:42:19 PM: "bootdev none" instead 

Jeremy Kerr (petitboot): 
6:42:35 PM: yeah, whatever the REST equivalent of 'bootdev none' is :-) 

dlshaw: 
6:42:41 PM: ok

williamspatrick, 2017-01-13:
I think this is just a matter of documentation.  We already have a translation between a settingsd, which is where the boot config is stored, and the ipmi message up to petitboot.  @ratagupt - I think you wrote this?

jk-ozlabs, 2017-01-16:
From @zmoose's description, it also sounds like the IPMI Set Chassis Boot Options command isn't fully functional. Petitboot uses this to provide the "Clear option" facility, which didn't seem to clear out the IPMI-specified boot option.

Petitboot's "clear" facility is equivalent to sending a `chassis bootdev none options=persistent` through inband IPMI.

@zmoose: you did try the "Clear option" button in the "IPMI boot option" section of petitboot config, is that correct?

zmoose, 2017-01-16:
@jk-ozlabs Correct I did the "Clear option" in Petitboot and then manually booted the host OS (Ubuntu).  I then issued a "reboot" from the Ubuntu OS as root and the system rebooted and halted again at Petitboot and the " Temporary IPMI boot option: Optical" was still showing.  I did this two more times and each time that was the case. 

However, the next day Mike Walsh had the same system and reloaded the firmware (same firmware we had before) and the "Clear option" did seem to work as I would expect since the reboot from Ubuntu host OS resulted in the system rebooting and Petiboot's autoboot starting the host OS (Ubuntu) automatically after 10 seconds. 

Though it seems whenever we load firmware the first boot always halts at Petitboot due to " Temporary IPMI boot option: Optical" so we need a method for our automation to disable this so Petitboot's autoboot works after loading new firmware.

jk-ozlabs, 2017-01-16:
@zmoose: ok, excellent. When you say you reloaded the firmware there, was that the host firmware, or the BMC firmware?

jk-ozlabs, 2017-01-16:
... and it looks like we'll need to ensure that the factory default is for no IPMI boot option.

zmoose, 2017-01-16:
@jk-ozlabs  the firmware we used was 920.1702.20170110a.  We load both BMC and PNOR.

PNOR levels in 920.1702.20170110a 
IBM-barreleye-ibm-OP8_v1.12_2.89
    op-build-7ab5c0c
    buildroot-51f1ac9
    skiboot-5.4.2
    hostboot-e28b28f
    linux-4.4.39-openpower1-24b42f6
    petitboot-v1.4.0-eab7ae1
    barreleye-xml-253622f
    occ-d7efe30
    hostboot-binaries-464ab6a
    capp-ucode-1bb7503

zmoose, 2017-02-01:
This issue was caused by another teams test automation leaving the system boot device set to CDROM.  This has been corrected with https://gerrit.openbmc-project.xyz/#/c/2204/2

So closing this issue.

closed 2017-02-01 by zmoose