openbmc/openbmc #966: [Witherspoon - v1.99.0-383-g4a2201f] Driver 442: Hit with error 503-Intermittent Service Unavailable during GET access of chassis/attr/present

sivassrr, 2017-01-14:
With driver 442, Tested entire Witherspoon bucket 10 times - and found 2 time TC (Test Case) which is 
related to GET chassis present failed.  Failure details as follows:
```
Send a GET request on the session object found using the
Start / End / Elapsed: 	20170113 10:54:48.217 / 20170113 10:54:48.730 / 00:00:00.513
10:54:48.729 	INFO 	Get Request using : alias=openbmc, uri=/org/openbmc/inventory/system/chassis/attr/present, headers=None 	
10:54:48.729 	INFO 	${ret} = <Response [503]> 	
00:00:00.011 KEYWORD BuiltIn . Run Keyword If '${quiet}' == '${0}', Log Response, ${ret}
Documentation: 	

Runs the given keyword with the given arguments, if condition is true.
Start / End / Elapsed: 	20170113 10:54:48.730 / 20170113 10:54:48.741 / 00:00:00.011
00:00:00.003 KEYWORD rest_client . Log Response ${ret}
Start / End / Elapsed: 	20170113 10:54:48.738 / 20170113 10:54:48.741 / 00:00:00.003
00:00:00.000 KEYWORD ${msg} = BuiltIn . Catenate SEPARATOR=, Response code:, ${resp.status_code}, , Content:, ${resp.content}
Documentation: 	

Catenates the given items together and returns the resulted string.
Start / End / Elapsed: 	20170113 10:54:48.739 / 20170113 10:54:48.739 / 00:00:00.000
10:54:48.739 	INFO 	${msg} = Response code:503, Content:{
  "data": {
    "description": "org.freedesktop.DBus.Error.ObjectPathInUse: busy processing bus traffic"
  }, 
  "message": "503 Service Unavailable", 
  "status": "error".
```
[sensorchassispresent.zip](https://github.com/openbmc/openbmc/files/705946/sensorchassispresent.zip)

sivassrr, 2017-01-14:
For the entire test report of Driver 442, Please check https://github.com/openbmc/openbmc-test-automation/issues/220

williamspatrick, 2017-03-10:
This has been fixed in the latest code.

closed 2017-03-10 by williamspatrick