openbmc/openbmc #967: [Witherspoon - v1.99.0-383-g4a2201f] Driver 442: Hit with error 503-Intermittent Service Unavailable during POST call of event acceptTestMessage

sivassrr, 2017-01-14:
With driver 442, Tested Witherspoon bucket 10 times - and found 1 time TC failed (Test Case) which is
related to POST REST call of event acceptTestMessage . Failure details as follows:
```
KEYWORD ${ret} = RequestsLibrary.RequestsKeywords . Post Request openbmc, ${base_uri}, &{kwargs}, timeout=${timeout}
Documentation: 	

Send a POST request on the session object found using the
Start / End / Elapsed: 	20170113 11:20:23.403 / 20170113 11:20:23.855 / 00:00:00.452
11:20:23.854 	INFO 	Post Request using : alias=openbmc, uri=/org/openbmc/records/events/action/acceptTestMessage, data={"data": []}, headers={u'Content-Type': u'application/json'}, files=None, allow_redirects=True 	
11:20:23.855 	INFO 	${ret} = <Response [503]> 	
00:00:00.015 KEYWORD BuiltIn . Run Keyword If '${quiet}' == '${0}', Log Response, ${ret}
Documentation: 	

Runs the given keyword with the given arguments, if condition is true.
Start / End / Elapsed: 	20170113 11:20:23.855 / 20170113 11:20:23.870 / 00:00:00.015
00:00:00.007 KEYWORD rest_client . Log Response ${ret}
Start / End / Elapsed: 	20170113 11:20:23.863 / 20170113 11:20:23.870 / 00:00:00.007
00:00:00.001 KEYWORD ${msg} = BuiltIn . Catenate SEPARATOR=, Response code:, ${resp.status_code}, , Content:, ${resp.content}
00:00:00.003 KEYWORD rest_client . Logging ${msg}, console=True
Start / End / Elapsed: 	20170113 11:20:23.867 / 20170113 11:20:23.870 / 00:00:00.003
00:00:00.001 KEYWORD BuiltIn . Log ${msg}, console=True
Documentation: 	

Logs the given message with the given level.
Start / End / Elapsed: 	20170113 11:20:23.868 / 20170113 11:20:23.869 / 00:00:00.001
11:20:23.869 	INFO 	Response code:503, Content:{
  "data": {
    "description": "org.freedesktop.DBus.Error.ObjectPathInUse: busy processing bus traffic"
  }, 
  "message": "503 Service Unavailable", 
  "status": "error"
```
[TestEventLogAcceptMessage.zip](https://github.com/openbmc/openbmc/files/705949/TestEventLogAcceptMessage.zip)

sivassrr, 2017-01-14:
For the entire test report of Driver 442, Please check https://github.com/openbmc/openbmc-test-automation/issues/220

williamspatrick, 2017-03-10:
This has been fixed in the latest code.

closed 2017-03-10 by williamspatrick