openbmc/openbmc #968: [Witherspoon - v1.99.0-383-g4a2201f] Driver 442: Hit with error 503-Intermittent Service Unavailable during REST GET request of URI /org/openbmc/settings/host0/attr/boot_flags

sivassrr, 2017-01-14:
With driver 442, Tested entire Witherspoon bucket 10 times - and found 1 time TC (Test Case) which is
related to "Set the boot device as network using IPMItool" failed. Using IPMITool setting the value and verifying the set using REST interface using the URI /org/openbmc/settings/host0/attr/boot_flags.
[setbootnetwork.zip](https://github.com/openbmc/openbmc/files/705953/setbootnetwork.zip)

```
Documentation: 	
Send a GET request on the session object found using the
Start / End / Elapsed: 	20170113 11:17:23.024 / 20170113 11:17:23.384 / 00:00:00.360
11:17:23.383 	INFO 	Get Request using : alias=openbmc, uri=/org/openbmc/settings/host0/attr/boot_flags, headers=None 	
11:17:23.384 	INFO 	${ret} = <Response [503]> 	
00:00:00.008 KEYWORD BuiltIn . Run Keyword If '${quiet}' == '${0}', Log Response, ${ret}
Documentation: 	

Runs the given keyword with the given arguments, if condition is true.
Start / End / Elapsed: 	20170113 11:17:23.384 / 20170113 11:17:23.392 / 00:00:00.008
00:00:00.004 KEYWORD rest_client . Log Response ${ret}
Start / End / Elapsed: 	20170113 11:17:23.388 / 20170113 11:17:23.392 / 00:00:00.004
00:00:00.001 KEYWORD ${msg} = BuiltIn . Catenate SEPARATOR=, Response code:, ${resp.status_code}, , Content:, ${resp.content}
Documentation: 	

Catenates the given items together and returns the resulted string.
Start / End / Elapsed: 	20170113 11:17:23.389 / 20170113 11:17:23.390 / 00:00:00.001
11:17:23.390 	INFO 	${msg} = Response code:503, Content:{
  "data": {
    "description": "org.freedesktop.DBus.Error.ObjectPathInUse: busy processing bus traffic"
  }, 
  "message": "503 Service Unavailable", 
  "status": "error"...
```

sivassrr, 2017-01-14:
For the entire test report of Driver 442, Please check https://github.com/openbmc/openbmc-test-automation/issues/220

williamspatrick, 2017-03-10:
This has been fixed in the latest code.

closed 2017-03-10 by williamspatrick