openbmc/openbmc #969: [Witherspoon - v1.99.0-383-g4a2201f] Driver 442: Hit with error 503-Intermittent Service Unavailable during PUT access of URI /org/openbmc/settings/host0/attr/power_cap

sivassrr, 2017-01-14:
With driver 442, Tested entire Witherspoon bucket 10 times - and found 2 time TC failed (Test Case) which is related to "Set the power with MIN_POWER value". 

```
KEYWORD ${ret} = RequestsLibrary.RequestsKeywords . Put Request openbmc, ${base_uri}, &{kwargs}, timeout=${timeout}
Documentation: 	

Send a PUT request on the session object found using the
Start / End / Elapsed: 	20170113 12:08:33.140 / 20170113 12:08:33.461 / 00:00:00.321
12:08:33.461 	INFO 	Put Request using : alias=openbmc, uri=/org/openbmc/settings/host0/attr/power_cap, data={"data": 0},                     headers={u'Content-Type': u'application/json'}, allow_redirects=True 	
12:08:33.461 	INFO 	${ret} = <Response [503]> 	
00:00:00.006 KEYWORD rest_client . Log Response ${ret}
Start / End / Elapsed: 	20170113 12:08:33.462 / 20170113 12:08:33.468 / 00:00:00.006
00:00:00.002 KEYWORD ${msg} = BuiltIn . Catenate SEPARATOR=, Response code:, ${resp.status_code}, , Content:, ${resp.content}
Documentation: 	

Catenates the given items together and returns the resulted string.
Start / End / Elapsed: 	20170113 12:08:33.463 / 20170113 12:08:33.465 / 00:00:00.002
12:08:33.465 	INFO 	${msg} = Response code:503, Content:{
  "data": {
    "description": "org.freedesktop.DBus.Error.ObjectPathInUse: busy processing bus traffic"
  }, 
  "message": "503 Service Unavailable", 
  "status": "error"...
```

[setpowerwithminpowervalue.zip](https://github.com/openbmc/openbmc/files/705957/setpowerwithminpowervalue.zip)

sivassrr, 2017-01-14:
For the entire test report of Driver 442, Please check https://github.com/openbmc/openbmc-test-automation/issues/220

sivassrr, 2017-01-14:
root@witherspoon:# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.0-383"
VERSION_ID="v1.99.0-383-g4a2201f-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.0-383"
BUILD_ID="v1.99.0"
root@witherspoon:#

vishwabmc, 2017-01-18:
Looking at the trend, this is happening in random usecases.. That makes me think that its something related to either (1) rest-dbus (2) mapper.

@bradbishop You have any idea ?

williamspatrick, 2017-03-10:
This has been fixed in the latest code.

closed 2017-03-10 by williamspatrick