openbmc/openbmc #972: [Barrelleye v1.99.0-385-ge9ba289]  Logging service multiple restarts

gkeishin, 2017-01-16:
Seen this on the newer driver.. a poweron resulted into Logging service to restart multiple time and failed

```
Nov 21 06:40:06 barreleye systemd[1]: xyz.openbmc_project.Logging.service: Start operation timed out. Terminating.
Nov 21 06:40:06 barreleye systemd[1]: Failed to start Phosphor Log Manager.
Nov 21 06:40:06 barreleye systemd[1]: xyz.openbmc_project.Logging.service: Unit entered failed state.
Nov 21 06:40:06 barreleye systemd[1]: xyz.openbmc_project.Logging.service: Failed with result 'timeout'.
Nov 21 06:40:07 barreleye systemd[1]: xyz.openbmc_project.Logging.service: Service hold-off time over, scheduling restart.
Nov 21 06:40:07 barreleye systemd[1]: Stopped Phosphor Log Manager.
```

- [x]  Journal log attached  
[journald_v1.99.0-385-ge9ba289_Logging.txt](https://github.com/openbmc/openbmc/files/707392/journald_v1.99.0-385-ge9ba289_Logging.txt)

bradbishop, 2017-01-17:
This should be fixed by https://gerrit.openbmc-project.xyz/#/c/1788/

gkeishin, 2017-01-17:
great.. thanks @bradbishop

gkeishin, 2017-02-21:
Closing as verified on the latter builds,, Thanks

closed 2017-02-21 by gkeishin