openbmc/openbmc #973: [Witherspoon - v1.99.0-385-ge9ba289] Driver 443: Hit with error 503-Intermittent Service Unavailable during set manual time mode

sivassrr, 2017-01-16:
Driver 442 we didnt hit with this error or any time related issue https://github.com/openbmc/openbmc/issues/915.

But from Driver 443 of openpower.xyz we started seeing 503-Service Unavailable instead of dbus exception. 

``` 
Details as follows:
Full Name: 	witherspoonCombinedReport.Tests.Test Time.Set Manual Time Mode
Documentation: 	

**GOOD PATH** This testcase is to set time mode as manual using REST URI and then verify using REST API.
Tags: 	Set_Manual_Time_Mode
Start / End / Elapsed: 	20170115 19:03:32.807 / 20170115 19:04:18.094 / 00:00:45.287
Status: 	FAIL (critical)
Message: 	{u'description': u'org.freedesktop.DBus.Error.ObjectPathInUse: busy processing bus traffic'} != Manual
00:00:01.618 KEYWORD Set Time Mode Manual
00:00:02.423 KEYWORD ${boot} = rest_client . Read Attribute ${SETTING_HOST}, time_mode
Start / End / Elapsed: 	20170115 19:03:34.426 / 20170115 19:03:36.849 / 00:00:02.423
00:00:02.420 KEYWORD ${resp} = rest_client . OpenBMC Get Request ${uri}/attr/${attr}, timeout=${timeout}, quiet=${quiet}
00:00:00.001 KEYWORD ${content} = RequestsLibrary.RequestsKeywords . To Json ${resp.content}
Documentation: 	

Convert a string to a JSON object
Start / End / Elapsed: 	20170115 19:03:36.847 / 20170115 19:03:36.848 / 00:00:00.001
19:03:36.848 	INFO 	To JSON using : content={
  "data": {
    "description": "org.freedesktop.DBus.Error.ObjectPathInUse: busy processing bus traffic"
  }, 
  "message": "503 Service Unavailable", 
  "status": "error"
} 	
19:03:36.848 	INFO 	To JSON using : pretty_print=False 	
19:03:36.848 	INFO 	${content} = {u'message': u'503 Service Unavailable', u'status': u'error', u'data': {u'description': u'org.freedesktop.DBus.Error.ObjectPathInUse: busy processing bus traffic'}} 	
19:03:36.849 	INFO 	${boot} = {u'description': u'org.freedesktop.DBus.Error.ObjectPathInUse: busy processing bus traffic'}
```

sivassrr, 2017-01-16:
Details about BMC version as follows:
```
root@witherspoon:# cat /etc/version
20170115005748
root@witherspoon:# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.0-385"
VERSION_ID="v1.99.0-385-ge9ba289-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.0-385"
BUILD_ID="v1.99.0"
```

sivassrr, 2017-01-16:
[SetManualTimeMode.zip](https://github.com/openbmc/openbmc/files/708570/SetManualTimeMode.zip)
FFDC File is attached for this new failure while setting manual time mode.

vishwabmc, 2017-01-18:
I checked journal and saw that timemanager was all happier. Need to see if this request did even get to time manager -or- this error was thrown at rest-dbus layer ?

@sivassrr How often have you seen this ?. 

Was this seen only when setting the time mode ?. 

How can I recreate this ?

sivassrr, 2017-01-18:
Yes, Seen this error only during setting the time mode.

@vishwabmc  https://github.com/openbmc/openbmc-test-automation/issues/220 last update has Driver 449 report.  There I didnt see this error. But the other TC "Set time owner as BMC" is failing 2 times. 

I believe we may want to open separate issue for that. Currently I am trying with the latest driver. Will check and open it accordingly.

sivassrr, 2017-01-18:
Earlier same test case was failing with "Dbus exception error" now it is become 503-Service unavailable error. Let me watch for latest driver.

sivassrr, 2017-01-19:
Same with Driver 453 as well

sivassrr, 2017-01-19:
Recreate:  Run loop test of either entire bucket or only Time Suite, we can observe this.

williamspatrick, 2017-03-10:
This has been fixed in the latest code.

closed 2017-03-10 by williamspatrick