openbmc/openbmc #974: Add delay on retries in mapper on org.freedesktop.DBus.Error.ObjectPathInUse exception

geissonator, 2017-01-16:
See https://gerrit.openbmc-project.xyz/#/c/1751/1/obmc/mapper/bindings.py@55 for the code review comment.

See #862 for the issue we're trying to fix.

The goal of this issue is to change the defaults in the mapper code to allow for a delay when a retry is needed so that we stop hammering the dbus.

williamspatrick, 2017-01-17:
@cklokma - This is the one we were thinking of starting you with.

vishwabmc, 2017-01-18:
@williamspatrick : Sivas opened multiple issues on the similar failures. Are you saying this issue will be addressed by @cklokma  and I can reference this issue in all those that Sivass opened ?

geissonator, 2017-01-18:
@vishwabmc we're not positive this will clear up the issue Sivas is seeing but it will definitely reduce the amount of dbus transactions we're doing now.

closed 2017-01-25 by williamspatrick