openbmc/openbmc #975: 1w backtrace on Zaius

shenki, 2017-01-17:
Mikey saw this on a Zaius BMC. Are we adding the same thing twice? Or is this an issue in the driver that needs addressing?

```
------------[ cut here ]------------
WARNING: CPU: 0 PID: 353 at /var/lib/jenkins/workspace/openbmc-build-gerrit-trigger-multi/distro/ubuntu/target/zaius/openbmc/build/tmp/work-shared/zaius/kernel-source/fs/sysfs/dir.c:31 sysfs_warn_dup+0x50/0x70
sysfs: cannot create duplicate filename '/bus/w1/devices/00-800000000000'
CPU: 0 PID: 353 Comm: w1_bus_master3 Not tainted 4.7.10-120956cab11d86fa801d53811ed2f7f844216a2f #1
Hardware name: ASpeed SoC
[<801072f8>] (unwind_backtrace) from [<80105558>] (show_stack+0x10/0x14)
[<80105558>] (show_stack) from [<80110cc0>] (__warn+0xc4/0xf0)
[<80110cc0>] (__warn) from [<80110d30>] (warn_slowpath_fmt+0x44/0x6c)
[<80110d30>] (warn_slowpath_fmt) from [<801f8734>] (sysfs_warn_dup+0x50/0x70)
[<801f8734>] (sysfs_warn_dup) from [<801f8980>] (sysfs_do_create_link_sd+0x7c/0x94)
[<801f8980>] (sysfs_do_create_link_sd) from [<802b34d0>] (bus_add_device+0xe0/0x17c)
[<802b34d0>] (bus_add_device) from [<802b23b0>] (device_add+0x284/0x53c)
[<802b23b0>] (device_add) from [<802e9454>] (w1_attach_slave_device+0x204/0x3a0)
[<802e9454>] (w1_attach_slave_device) from [<802e9738>] (w1_slave_found+0x98/0xd0)
[<802e9738>] (w1_slave_found) from [<802e9f88>] (w1_search+0x1f0/0x280)
[<802e9f88>] (w1_search) from [<802ea084>] (w1_search_process_cb+0x6c/0x108)
[<802ea084>] (w1_search_process_cb) from [<802ea1ec>] (w1_process+0x68/0x114)
[<802ea1ec>] (w1_process) from [<801268dc>] (kthread+0xdc/0xf4)
[<801268dc>] (kthread) from [<80102278>] (ret_from_fork+0x14/0x3c)
---[ end trace 9e3a846ee0859652 ]---
w1_slave_driver 00-800000000000: Device registration [00-800000000000] failed. err=-17
w1_master_driver w1_bus_master3: w1_attach_slave_device: Attaching 00-800000000000 failed.
w1_master_driver w1_bus_master4: Family 0 for 00.800000000000.8c is not registered.
------------[ cut here ]------------
WARNING: CPU: 0 PID: 355 at /var/lib/jenkins/workspace/openbmc-build-gerrit-trigger-multi/distro/ubuntu/target/zaius/openbmc/build/tmp/work-shared/zaius/kernel-source/fs/sysfs/dir.c:31 sysfs_warn_dup+0x50/0x70
sysfs: cannot create duplicate filename '/bus/w1/devices/00-800000000000'
CPU: 0 PID: 355 Comm: w1_bus_master4 Tainted: G        W       4.7.10-120956cab11d86fa801d53811ed2f7f844216a2f #1
Hardware name: ASpeed SoC
[<801072f8>] (unwind_backtrace) from [<80105558>] (show_stack+0x10/0x14)
[<80105558>] (show_stack) from [<80110cc0>] (__warn+0xc4/0xf0)
[<80110cc0>] (__warn) from [<80110d30>] (warn_slowpath_fmt+0x44/0x6c)
[<80110d30>] (warn_slowpath_fmt) from [<801f8734>] (sysfs_warn_dup+0x50/0x70)
[<801f8734>] (sysfs_warn_dup) from [<801f8980>] (sysfs_do_create_link_sd+0x7c/0x94)
[<801f8980>] (sysfs_do_create_link_sd) from [<802b34d0>] (bus_add_device+0xe0/0x17c)
[<802b34d0>] (bus_add_device) from [<802b23b0>] (device_add+0x284/0x53c)
[<802b23b0>] (device_add) from [<802e9454>] (w1_attach_slave_device+0x204/0x3a0)
[<802e9454>] (w1_attach_slave_device) from [<802e9738>] (w1_slave_found+0x98/0xd0)
[<802e9738>] (w1_slave_found) from [<802e9f88>] (w1_search+0x1f0/0x280)
[<802e9f88>] (w1_search) from [<802ea084>] (w1_search_process_cb+0x6c/0x108)
[<802ea084>] (w1_search_process_cb) from [<802ea1ec>] (w1_process+0x68/0x114)
[<802ea1ec>] (w1_process) from [<801268dc>] (kthread+0xdc/0xf4)
[<801268dc>] (kthread) from [<80102278>] (ret_from_fork+0x14/0x3c)
---[ end trace 9e3a846ee0859653 ]---
w1_slave_driver 00-800000000000: Device registration [00-800000000000] failed. err=-17
w1_master_driver w1_bus_master4: w1_attach_slave_device: Attaching 00-800000000000 failed.
w1_master_driver w1_bus_master1: Family 0 for 00.800000000000.8c is not registered.
------------[ cut here ]------------
```

GHF, 2017-01-17:
it's an issue with the w1 driver when the bus is stuck at a low state, which is what happens on EVT1 boards when no thermal sensors are plugged in (the 1-wire bus pull up resistors are on the sensor board). When the driver probes, it thinks that a slave device is responding, so it keeps creating slave devices with the same addresses.

There are 100k pull up resistors on 1-wire buses on >EVT2 boards, so this is fixed on EVT2 after applying https://github.com/openbmc/linux/commit/2499942f10b161f6c980e558ffa3f99402b1710e

GHF, 2017-01-19:
correction: the necessary resistors are present on *>=EVT2* boards. They can also be stuffed onto EVT1 boards if that's all that you have.

gtmills, 2018-07-09:
@shenki @GHF Still a problem?

GHF, 2018-07-09:
Fixed in hardware.

closed 2018-07-09 by gtmills