openbmc/openbmc #978: [ Barreleye v1.99.0-392-g8e04907 ] DBusException: org.freedesktop.DBus.Error.LimitsExceeded:

gkeishin, 2017-01-17:
Build https://openpower.xyz/job/openbmc-build/449/

Just tried reboot use case from BMC standby and its stuck at BMC starting

Nov 21 06:09:55 barreleye system_manager.py[740]: Traceback (most recent call last):
Nov 21 06:09:55 barreleye system_manager.py[740]:   File "/usr/lib/python2.7/site-packages/dbus/connection.py", line 607, in msg_reply_handler
Nov 21 06:09:55 barreleye system_manager.py[740]:     *message.get_args_list()))
Nov 21 06:09:55 barreleye system_manager.py[740]:   File "/usr/lib/python2.7/site-packages/obmc/mapper/utils.py", line 93, in <lambda>
Nov 21 06:09:55 barreleye system_manager.py[740]:     path, retry, x))
Nov 21 06:09:55 barreleye system_manager.py[740]:   File "/usr/lib/python2.7/site-packages/obmc/mapper/utils.py", line 114, in get_object_error
Nov 21 06:09:55 barreleye system_manager.py[740]:     self.error_callback(e)
Nov 21 06:09:55 barreleye system_manager.py[740]:   File "/usr/lib/python2.7/site-packages/obmc/mapper/utils.py", line 56, in default_error
Nov 21 06:09:55 barreleye system_manager.py[740]:     raise e
```
Nov 21 06:09:55 barreleye system_manager.py[740]: dbus.exceptions.DBusException: org.freedesktop.DBus.Error.LimitsExceeded: The maximum number of pending replies per connection has been reached
```

- [x] Journal Log - 
[journald_v1.99.0-392-g8e04907_BMC_STARTING.txt](https://github.com/openbmc/openbmc/files/711153/journald_v1.99.0-392-g8e04907_BMC_STARTING.txt)

vishwabmc, 2017-01-18:
Looks like we have started to stress the dbus infrastructure.

Can I get access to the box ?

geissonator, 2017-01-19:
Lei recently put some retry code in on the org.freedesktop.DBus.Error.ObjectPathInUse errors.  For now, we should probably look around for where that is captured and retried and add in this new org.freedesktop.DBus.Error.LimitsExceeded.  Also, #974 may help as well.

gkeishin, 2017-01-20:
@vishwabmc  We hit this again today with newer driver v1.99.0-421-g48b42e0. 

- New journal log for reference  
[journald_v1.99.0-421-g48b42e0.DBus.Error.LimitsExceeded.txt](https://github.com/openbmc/openbmc/files/718611/journald_v1.99.0-421-g48b42e0.DBus.Error.LimitsExceeded.txt)

spinler, 2017-01-27:
Issue 974 went into obmc with 36f0ff09e on 1/25.  
@geissonator @gkeishin has this been hit since then?

spinler, 2017-01-27:
Making the changes to have the same retries occur on LimitsExceeded as ObjectPath InUse

spinler, 2017-01-27:
Even with the latest code, both before and after my fix I would still see occasional ObjectPathInUse errors.  I haven't seen any LimitsExceeded errors with my change applied.

spinler, 2017-01-27:
https://gerrit.openbmc-project.xyz/#/c/2107/

closed 2017-01-30 by williamspatrick