openbmc/openbmc #981: Yocto toolchain should use target kernel headers for builds

amboar, 2017-01-18:
The `mboxd` tool requires kernel headers from the target (openbmc) kernel, but these are not the headers the toolchain has access to when building userspace programs. In this case (where the necessary header isn't included in the pre-packaged headers) the only work-around is to patch the kernel header into the userspace source, which is prone to getting out of sync and breaking.

Fix Yocto so the headers from the target kernel are installed for use by the toolchain.

kc8apf, 2017-01-18:
See the comment at the top of
http://git.yoctoproject.org/cgit.cgi/poky/plain/meta/recipes-kernel/linux-libc-headers/linux-libc-headers.inc

williamspatrick, 2017-01-19:
The recommendation in that file doesn't work in this case. You can't include headers for userspace from that location because you end up with a compiler warning. (Linux/types.h has a #warning in the kernel tree).

kc8apf, 2017-01-19:
When I look in STAGING_KERNEL_DIR
(build/tmp/work-shared/${MACHINE}/kernel-source), include/linux/types.h has
no #warning.  It does #include <uapi/linux/types.h> which will do a
#warning if __EXPORTED_HEADERS__ isn't set but include/linux/types.h does
set that on the line before the include.

kc8apf, 2017-01-19:
Or if you really need to, in your recipe, run 'make headers_install' from
STAGING_KERNEL_DIR and have it install to a path within your recipe's work
directory.

amboar, 2017-01-19:
> When I look in STAGING_KERNEL_DIR (build/tmp/work-shared/${MACHINE}/kernel-source), include/linux/types.h has no #warning. It does #include <uapi/linux/types.h> which will do a #warning if __EXPORTED_HEADERS__ isn't set but include/linux/types.h does set that on the line before the include.

That doesn't gel with the fact that the build breaks for me and @williamspatrick . Do you have a patch/build log demonstrating it works?

> Or if you really need to, in your recipe, run 'make headers_install' from STAGING_KERNEL_DIR and have it install to a path within your recipe's work directory.

That is a less brittle solution but feels like a hack and isn't very scalable. Why not just do it right once to accommodate all packages?

kc8apf, 2017-01-19:
I don't have a project that uses kernel headers on hand. Have you looked at
other projects that have this problem? What do they do?

Other than upstream Yocto developers explicitly saying it's the wrong
approach in the recipe that would need to change? If nothing else, ask what
to do on the Yocto mailing list.

peteut, 2019-02-28:
You're probably looking for `make -C $KDIR prepare_modules`. This is required when you're building out of tree modules.

mdmillerii, 2019-03-11:
I would guess this is not default because it makes the toolchain (and hence all binaries) machine specific.

closed 2019-10-10 by stale[bot]