openbmc/openbmc #982: [witherspoon v1.99.0-394-g9dcf2cb] Error log events are not cleared or deleted

gkeishin, 2017-01-18:
The services seems to up and running, even restarting doesnt seem to solve this. 

root@witherspoon:~# systemctl status org.openbmc.records.events.service | cat
● org.openbmc.records.events.service - Phosphor Event Manager
   Loaded: loaded (/lib/systemd/system/org.openbmc.records.events.service; enabled; vendor preset: enabled)
   Active: active (running) since Mon 2016-11-21 06:08:19 UTC; 28min ago
 Main PID: 725 (phosphor-eventd)
   CGroup: /system.slice/org.openbmc.records.events.service
           └─725 /usr/sbin/phosphor-eventd -s ${LOG_SIZE_LIMIT} -t ${LOG_COUNT_LIMIT}

Nov 21 06:08:14 witherspoon systemd[1]: Starting Phosphor Event Manager...
Nov 21 06:08:20 witherspoon phosphor-eventd[725]: 0
Nov 21 06:08:19 witherspoon systemd[1]: Started Phosphor Event Manager.

The failure for Clearing the event and deleting are seen on test as well. Just an illustration in CURL to show it didn't.. The journal log but i dont see any suspicious info there --> here 
[journald_v1.99.0-394-g9dcf2cb_error_event.txt](https://github.com/openbmc/openbmc/files/713144/journald_v1.99.0-394-g9dcf2cb_error_event.txt)

$curl -c cjar -b cjar -k -H "Content-Type: application/json" -X POST https://xx.xx.xx.xx/org/openbmc/records/events/action/clear -d "{\"data\": []}"
{
  "data": 0, 
  "message": "200 OK", 
  "status": "ok"
$  curl -c cjar -b cjar -k -H "Content-Type: application/json" -X GET https://xx.xx.xx.xx/org/openbmc/records/events
{
  "data": [
    "/org/openbmc/records/events/1019", 
    "/org/openbmc/records/events/1008", 
    "/org/openbmc/records/events/1014", 
    "/org/openbmc/records/events/1013", 
    "/org/openbmc/records/events/1010", 
    "/org/openbmc/records/events/1011", 
    "/org/openbmc/records/events/1001", 
    "/org/openbmc/records/events/1000", 
    "/org/openbmc/records/events/1007", 
    "/org/openbmc/records/events/1005", 
    "/org/openbmc/records/events/1017", 
    "/org/openbmc/records/events/1020", 
    "/org/openbmc/records/events/1022", 
    "/org/openbmc/records/events/1002", 
    "/org/openbmc/records/events/1012"
  ], 
  "message": "200 OK", 
  "status": "ok"
}

gkeishin, 2017-01-18:
@anoo1 Anything on this.. This seems to specific to Witherspoon but not seen barreleye..

vishwabmc, 2017-01-18:
The events should get persisted on some /var/lib/obmc ... Can I get on the box ?

gkeishin, 2017-01-19:
@vishwabmc You can take a look w6.. its in same error state like reported earlier here.

Response code:500, Content:{
  "data": {
    "description": "Internal Server Error", 
    "exception": "DBusException(\"Unknown interface 'org.openbmc.Object.Delete'.\",)", 
    "traceback": [
      "Traceback (most recent call last):", 
      "  File \"/usr/lib/python2.7/site-packages/bottle.py\", line 862, in _handle", 
      "    return route.call(**args)", 
      "  File \"/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py\", line 522, in __call__", 
      "    return self.callback(*a, **kw)", 
      "  File \"/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py\", line 678, in wrap", 
      "    return self.to_jsonp(callback(*a, **kw))", 
      "  File \"/usr/lib/python2.7/site-packages/bottle.py\", line 1734, in wrapper", 
      "    rv = callback(*a, **ka)", 
      "  File \"/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py\", line 613, in wrap", 
      "    resp = {'data': callback(*a, **kw)}", 
      "  File \"/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py\", line 576, in wrap", 
      "    return callback(*a, **kw)", 
      "  File \"/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py\", line 601, in wrap", 
      "    return callback(*a, **kw)", 
      "  File \"/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py\", line 87, in __call__", 
      "    return getattr(self, 'do_' + request.method.lower())(**kw)", 
      "  File \"/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py\", line 369, in do_get", 
      "    mapper_data=request.route_data['map'])", 
      "  File \"/usr/lib/python2.7/site-packages/obmc/mapper/bindings.py\", line 126, in enumerate_object", 
      "    path, owner, interfaces, match))", 
      "  File \"/usr/lib/python2.7/site-packages/obmc/mapper/bindings.py\", line 110, in __get_properties_on_bus", 
      "    properties_iface, i))", 
      "  File \"/usr/lib/python2.7/site-packages/obmc/mapper/bindings.py\", line 96, in __get_properties_on_iface", 
      "    properties_iface.GetAll, iface)", 
      "  File \"/usr/lib/python2.7/site-packages/obmc/mapper/bindings.py\", line 82, in __try_properties_interface", 
      "    return f(*a)", 
      "  File \"/usr/lib/python2.7/site-packages/dbus/proxies.py\", line 145, in __call__", 
      "    **keywords)", 
      "  File \"/usr/lib/python2.7/site-packages/dbus/connection.py\", line 651, in call_blocking", 
      "    message, timeout)", 
      "DBusException: org.freedesktop.DBus.Error.UnknownInterface: Unknown interface 'org.openbmc.Object.Delete'."
    ]
  }, 
  "message": "500 Internal Server Error", 
  "status": "error"

vishwabmc, 2017-01-19:
I got on the box and observed that the `org.openbmc.Delete` on the individual error events are getting deleted. However, the 'org.openbmc.recordlog.clear` is not working even with the direct `busctl` call.

Did some change go recently on this ?

@Josh - I see the latest commit is from you.. did you observe clear worked ?

vishwabmc, 2017-01-19:
@JoshDKing ^

vishwabmc, 2017-01-19:
@geissonator Should this be assigned to @JoshDKing

JoshDKing, 2017-01-19:
Hmm I don't remember anything like that, for what I was doing everything seemed fine.

gkeishin, 2017-01-20:
@JoshDKing @vishwabmc   I had a word with @anoo1 on this.. we are moving away from old interfaces. So she is making the changes but somehow witherspoon only seems to have impact because of the changes.

gkeishin, 2017-02-03:
Closing this since the infra structure is going in for this to xyz.

closed 2017-02-03 by gkeishin