openbmc/openbmc #983: SLP server logging too much data to the journal

geissonator, 2017-01-18:
Not sure if we need 2 issues here?  First issue is that the SLP server is doing a cout of all transactions, https://github.com/openbmc/slpd-lite/blob/309ac4458146338741ab1572a28c3ef3d81f3bb1/main.cpp#L33  This needs to be removed.

Second is we're getting a lot of errors with this enabled on our Zaius system in the lab:

```
Jan 18 15:51:00 zaius4 slpd[816]: SLP Response
Jan 18 15:51:00 zaius4 slpd[816]: 02 02 00 00 12 20 00 00 00 00 36 43 00 02 65 6e 0a e8 I> Channel::Read : DataIn Fd[5] Req[38] Recv[38]
Jan 18 15:51:00 zaius4 slpd[816]: SLP Request
Jan 18 15:51:00 zaius4 slpd[816]: 02 01 00 00 38 20 00 00 00 00 35 ae 00 02 65 6e 00 00 00 17 73 65 72 76 69 63 65 3a 64 69 72 65 63 74 6f 72 79 2d 61 67 65 6e 74 00 07 44 45 46 41 55 4c 54 00 00 00 00
Jan 18 15:51:00 zaius4 slpd[816]: E> SLP Error rc=10
Jan 18 15:51:01 zaius4 slpd[816]: SLP Response
Jan 18 15:51:01 zaius4 slpd[816]: 02 02 00 00 12 20 00 00 00 00 35 ae 00 02 65 6e 0a e8 I> Channel::Read : DataIn Fd[5] Req[38] Recv[38]
Jan 18 15:51:02 zaius4 slpd[816]: SLP Request
Jan 18 15:51:02 zaius4 slpd[816]: 02 01 00 00 38 20 00 00 00 00 05 44 00 02 65 6e 00 00 00 17 73 65 72 76 69 63 65 3a 64 69 72 65 63 74 6f 72 79 2d 61 67 65 6e 74 00 07 44 45 46 41 55 4c 54 00 00 00 00
Jan 18 15:51:02 zaius4 slpd[816]: E> SLP Error rc=10
Jan 18 15:51:02 zaius4 slpd[816]: SLP Response
Jan 18 15:51:02 zaius4 slpd[816]: 02 02 00 00 12 20 00 00 00 00 05 44 00 02 65 6e 0a e8 I> Channel::Read : DataIn Fd[5] Req[41] Recv[41]
Jan 18 15:51:02 zaius4 slpd[816]: SLP Request
Jan 18 15:51:02 zaius4 slpd[816]: 02 01 00 00 41 20 00 00 00 00 35 ae 00 02 65 6e 00 09 39 2e 33 2e 32 33 2e 31 33 00 17 73 65 72 76 69 63 65 3a 64 69 72 65 63 74 6f 72 79 2d 61 67 65 6e 74 00 07 44 45 46 41 55 4c 54 00 00 00 00
Jan 18 15:51:02 zaius4 slpd[816]: E> SLP Error rc=10
Jan 18 15:51:03 zaius4 slpd[816]: SLP Response
Jan 18 15:51:03 zaius4 slpd[816]: 02 02 00 00 12 20 00 00 00 00 35 ae 00 02 65 6e 0a e8 I> Channel::Read : DataIn Fd[5] Req[41] Recv[41]
Jan 18 15:51:04 zaius4 slpd[816]: SLP Request
Jan 18 15:51:04 zaius4 slpd[816]: 02 01 00 00 41 20 00 00 00 00 05 44 00 02 65 6e 00 09 39 2e 33 2e 32 33 2e 31 33 00 17 73 65 72 76 69 63 65 3a 64 69 72 65 63 74 6f 72 79 2d 61 67 65 6e 74 00 07 44 45 46 41 55 4c 54 00 00 00 00
Jan 18 15:51:04 zaius4 slpd[816]: E> SLP Error rc=10
Jan 18 15:51:40 zaius4 slpd[816]: SLP Response
Jan 18 15:51:40 zaius4 slpd[816]: 02 02 00 00 12 20 00 00 00 00 05 44 00 02 65 6e 0a e8 I> Channel::Read : DataIn Fd[5] Req[38] Recv[38]
Jan 18 15:51:40 zaius4 slpd[816]: SLP Request
Jan 18 15:51:40 zaius4 slpd[816]: 02 01 00 00 38 20 00 00 00 00 50 23 00 02 65 6e 00 00 00 17 73 65 72 76 69 63 65 3a 64 69 72 65 63 74 6f 72 79 2d 61 67 65 6e 74 00 07 44 45 46 41 55 4c 54 00 00 00 00
Jan 18 15:51:40 zaius4 slpd[816]: E> SLP Error rc=10
Jan 18 15:51:42 zaius4 slpd[816]: SLP Response
Jan 18 15:51:42 zaius4 slpd[816]: 02 02 00 00 12 20 00 00 00 00 50 23 00 02 65 6e 0a e8 I> Channel::Read : DataIn Fd[5] Req[41] Recv[41]
Jan 18 15:51:42 zaius4 slpd[816]: SLP Request
Jan 18 15:51:42 zaius4 slpd[816]: 02 01 00 00 41 20 00 00 00 00 50 23 00 02 65 6e 00 09 39 2e 33 2e 32 33 2e 31 33 00 17 73 65 72 76 69 63 65 3a 64 69 72 65 63 74 6f 72 79 2d 61 67 65 6e 74 00 07 44 45 46 41 55 4c 54 00 00 00 00
Jan 18 15:51:42 zaius4 slpd[816]: E> SLP Error rc=10
```

```
[zaius4]# cat /etc/os-release 
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.0-392"
VERSION_ID="v1.99.0-392-g8e04907-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.0-392"
BUILD_ID="v1.99.0"
```

ratagupt, 2017-01-19:
Is there a script which is sending too many SLP request to the server? I kept the cout there to aid the debugging.
For second issue I need the request details what is the requet data which is being sent?
eg slptool command with the data.

williamspatrick, 2017-01-19:
No one was running slptool.  This is a machine sitting idle in the lab and spewing this information.

geissonator, 2017-01-19:
The team did verify we don't see this on Witherspoon or Barreleye, not sure what's special about the Zaius setup.

ratagupt, 2017-01-19:
There would be slp command in the network.otherwise these logs can not come...We can change the network of the zaius machine and can verify it.

geissonator, 2017-01-24:
https://gerrit.openbmc-project.xyz/#/c/1999/ to disable this until we can get this issue figured out.

closed 2017-02-11 by williamspatrick