openbmc/openbmc #984: Invalid GPIO backtraces in journal on Zaius

spinler, 2017-01-18:
These backtraces are showing up on Zaius:
```
Jan 18 15:36:21 zaius4 kernel: ------------[ cut here ]------------
Jan 18 15:36:21 zaius4 kernel: WARNING: CPU: 0 PID: 860 at /var/lib/jenkins/workspace/openbmc-build/distro/ubuntu/target/zaius/openbmc/build/tmp/work-shared/zaius/kernel-source/drivers/gpio/gpiolib.c:100 gpio_to_desc+0x84/0x98
Jan 18 15:36:21 zaius4 kernel: invalid GPIO 528
Jan 18 15:36:21 zaius4 kernel: CPU: 0 PID: 860 Comm: button_reset.ex Not tainted 4.7.10-b40fcf1208470a9107caaf7e73387799f32c5111 #1
Jan 18 15:36:21 zaius4 kernel: Hardware name: ASpeed SoC
Jan 18 15:36:21 zaius4 kernel: [<801072f8>] (unwind_backtrace) from [<80105558>] (show_stack+0x10/0x14)
Jan 18 15:36:21 zaius4 kernel: [<80105558>] (show_stack) from [<80110cc0>] (__warn+0xc4/0xf0)
Jan 18 15:36:21 zaius4 kernel: [<80110cc0>] (__warn) from [<80110d30>] (warn_slowpath_fmt+0x44/0x6c)
Jan 18 15:36:21 zaius4 kernel: [<80110d30>] (warn_slowpath_fmt) from [<80294df8>] (gpio_to_desc+0x84/0x98)
Jan 18 15:36:21 zaius4 kernel: [<80294df8>] (gpio_to_desc) from [<80298c8c>] (export_store+0x38/0x118)
Jan 18 15:36:21 zaius4 kernel: [<80298c8c>] (export_store) from [<801f831c>] (kernfs_fop_write+0x140/0x1ac)
Jan 18 15:36:21 zaius4 kernel: [<801f831c>] (kernfs_fop_write) from [<801a3c08>] (__vfs_write+0x28/0x12c)
Jan 18 15:36:21 zaius4 kernel: [<801a3c08>] (__vfs_write) from [<801a3e80>] (vfs_write+0xb8/0x144)
Jan 18 15:36:21 zaius4 kernel: [<801a3e80>] (vfs_write) from [<801a400c>] (SyS_write+0x54/0xac)
Jan 18 15:36:21 zaius4 kernel: [<801a400c>] (SyS_write) from [<801021c0>] (ret_fast_syscall+0x0/0x3c)
Jan 18 15:36:21 zaius4 kernel: ---[ end trace a5a97e9f39fe74e0 ]---
Jan 18 15:36:21 zaius4 kernel: export_store: invalid GPIO 528
Jan 18 15:36:21 zaius4 kernel[777]: ------------[ cut here ]------------
```

@GHF Is this what you expect to happen when you said in https://gerrit.openbmc-project.xyz/#/c/1793  that this reset button GPIO won't work?

GHF, 2017-01-18:
Yes, GPIO 528 doesn't appear to Linux yet. It requires something like this patch: https://patchwork.ozlabs.org/patch/704921/

This isn't a fatal error and the GPIO export returns a useful errno, so it's really just the gpiolib code being overzealous with its WARN invocations.

shenki, 2017-01-19:
@amboar is working on adding the ast2500 pins to our GPIO driver

amboar, 2017-01-20:
I've pushed a branch that adds the GPIOs: https://github.com/amboar/linux/tree/aspeed-gpio

There are some caveats:

1. This will break the OpenBMC userspace as the GPIO offset value is dynamic, and changes when we add the extra IOs. It might be possible to work around this, I haven't looked too hard.
2. GPIOs Y on the AST2400 and at least GPIOs AC on the AST2500 aren't functional: Switching them to output and writing a value will not lead to a change in hardware state. It's not clear to me yet why this is the case.
3. Some changes for the AST2400's pinmux are provided to get GPIOs AA4-AB3 to function

GHF, 2017-01-20:
Thanks Andrew!

re (1): by "offset value," do you mean the number of the first GPIO? How should the userspace discover the offset? The Linux documentation suggests that it will be an attribute at `/sys/class/gpio/gpiochipN/base` with a value equal to N, "the first GPIO managed by this chip." I don't see it on my current image but I haven't tried your tree.

amboar, 2017-01-24:
@GHF Yes, the number of the first gpio. I don't know how it should be discovered; the `base` file seems redundant given you need to know N to access it. Obviously wildcards won't work well if we expose multiple gpiochips.

Separately, I've sent the patches upstream after testing on a Palmetto and the AST2500EVB:

* https://lkml.org/lkml/2017/1/23/12
* https://lkml.org/lkml/2017/1/23/15
* https://lkml.org/lkml/2017/1/23/16

closed 2017-09-18 by spinler