openbmc/openbmc #985: dbus-org.freedesktop.Avahi.service fails to start on Zaius

geissonator, 2017-01-18:
```
[zaius4]# cat /etc/os-release 
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.0-392"
VERSION_ID="v1.99.0-392-g8e04907-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.0-392"
BUILD_ID="v1.99.0"
```

```
[zaius4]# systemctl status dbus-org.freedesktop.Avahi.service | cat
\ufffd\ufffdavahi-daemon.service - Avahi mDNS/DNS-SD Stack
   Loaded: loaded (/lib/systemd/system/avahi-daemon.service; enabled; vendor preset: enabled)
   Active: active (running) since Wed 2017-01-18 15:30:58 UTC; 1h 42min ago
 Main PID: 759 (avahi-daemon)
   Status: "avahi-daemon 0.6.32 starting up."
   CGroup: /system.slice/avahi-daemon.service
           \ufffd\ufffd\ufffd\ufffd\ufffd\ufffd759 avahi-daemon: running [zaius4.local
           \ufffd\ufffd\ufffd\ufffd\ufffd\ufffd769 avahi-daemon: chroot helpe

Jan 18 15:31:05 zaius4 avahi-daemon[759]: Registering new address record for 9.3.23.13 on eth1.IPv4.
Jan 18 15:31:06 zaius4 avahi-daemon[759]: Joining mDNS multicast group on interface eth0.IPv6 with address fe80::a061:4eff:fe01:d772.
Jan 18 15:31:06 zaius4 avahi-daemon[759]: New relevant interface eth0.IPv6 for mDNS.
Jan 18 15:31:06 zaius4 avahi-daemon[759]: Registering new address record for fe80::a061:4eff:fe01:d772 on eth0.*.
Jan 18 15:31:12 zaius4 avahi-daemon[759]: Joining mDNS multicast group on interface eth1.IPv6 with address fe80::e03b:59ff:fe57:4ad5.
Jan 18 15:31:12 zaius4 avahi-daemon[759]: New relevant interface eth1.IPv6 for mDNS.
Jan 18 15:31:12 zaius4 avahi-daemon[759]: Registering new address record for fe80::e03b:59ff:fe57:4ad5 on eth1.*.
Jan 18 15:31:25 zaius4 avahi-daemon[769]: ../../avahi-0.6.32/avahi-daemon/chroot.c: open() failed: No such file or directory
Jan 18 15:31:25 zaius4 avahi-daemon[759]: Failed to open /usr/share/dbus-1/interfaces/org.freedesktop.Avahi.Server.xml: Invalid argument
Jan 18 15:31:25 zaius4 avahi-daemon[759]: Failed to load introspection data.
```

gkeishin, 2017-02-02:
Adding supplementary information for this.

SYSTEM     : w37
BUILD_ID   : https://openpower.xyz/job/openbmc-build/distro=ubuntu,target=witherspoon/502/
VERSION_ID : v1.99.1-81-g42a851a
BUILD TIME : 20170202033520
```
Jan 30 02:47:11 witherspoon avahi-daemon[743]: Found user 'avahi' (UID 999) and group 'avahi' (GID 999).
Jan 30 02:47:11 witherspoon avahi-daemon[743]: Successfully dropped root privileges.
Jan 30 02:47:12 witherspoon avahi-daemon[743]: avahi-daemon 0.6.32 starting up.
Jan 30 02:47:12 witherspoon avahi-daemon[743]: Successfully called chroot().
Jan 30 02:47:12 witherspoon avahi-daemon[743]: Successfully dropped remaining capabilities.
Jan 30 02:47:13 witherspoon avahi-daemon[743]: Loading service file /services/obmc_console.service.
Jan 30 02:47:13 witherspoon avahi-daemon[743]: Loading service file /services/phosphor_rest.service.
Jan 30 02:47:13 witherspoon avahi-daemon[743]: Loading service file /services/sftp-ssh.service.
Jan 30 02:47:13 witherspoon avahi-daemon[743]: Loading service file /services/sftp.service.
Jan 30 02:47:13 witherspoon avahi-daemon[743]: Loading service file /services/ssh.service.
Jan 30 02:47:13 witherspoon avahi-daemon[743]: Joining mDNS multicast group on interface eth0.IPv6 with address fe80::72e2:84ff:fe14:2045.
Jan 30 02:47:13 witherspoon avahi-daemon[743]: New relevant interface eth0.IPv6 for mDNS.
Jan 30 02:47:13 witherspoon avahi-daemon[743]: Joining mDNS multicast group on interface eth0.IPv4 with address 9.3.185.173.
Jan 30 02:47:13 witherspoon avahi-daemon[743]: New relevant interface eth0.IPv4 for mDNS.
Jan 30 02:47:13 witherspoon avahi-daemon[743]: Network interface enumeration completed.
Jan 30 02:47:13 witherspoon avahi-daemon[743]: Registering new address record for fe80::72e2:84ff:fe14:2045 on eth0.*.
Jan 30 02:47:13 witherspoon avahi-daemon[743]: Registering new address record for 9.3.185.173 on eth0.IPv4.
Jan 30 02:47:14 witherspoon avahi-daemon[743]: Server startup complete. Host name is witherspoon.local. Local service cookie is 2200599522.
Jan 30 02:47:15 witherspoon avahi-daemon[743]: Service "witherspoon" (/services/ssh.service) successfully established.
Jan 30 02:47:15 witherspoon avahi-daemon[743]: Service "sftp" (/services/sftp.service) successfully established.
Jan 30 02:47:15 witherspoon avahi-daemon[743]: Service "witherspoon" (/services/sftp-ssh.service) successfully established.
Jan 30 02:47:15 witherspoon avahi-daemon[743]: Service "phosphor_rest" (/services/phosphor_rest.service) successfully established.
Jan 30 02:47:15 witherspoon avahi-daemon[743]: Service "obmc_console" (/services/obmc_console.service) successfully established.
Jan 30 02:47:34 witherspoon avahi-daemon[780]: ../../avahi-0.6.32/avahi-daemon/chroot.c: open() failed: No such file or directory
Jan 30 02:47:34 witherspoon avahi-daemon[743]: Failed to open /usr/share/dbus-1/interfaces/org.freedesktop.Avahi.Server.xml: Invalid argument
Jan 30 02:47:34 witherspoon avahi-daemon[743]: Failed to load introspection data.
```
- [x] Journal Log  
[journald_v1.99.1-81-g42a851a_Avahi.txt](https://github.com/openbmc/openbmc/files/746944/journald_v1.99.1-81-g42a851a_Avahi.txt)

root@witherspoon:~# systemctl status dbus-org.freedesktop.Avahi.service | cat
● avahi-daemon.service - Avahi mDNS/DNS-SD Stack
   Loaded: loaded (/lib/systemd/system/avahi-daemon.service; enabled; vendor preset: enabled)
   Active: active (running) since Mon 2017-01-30 02:47:13 UTC; 3min 48s ago
 Main PID: 743 (avahi-daemon)
   Status: "avahi-daemon 0.6.32 starting up."
   CGroup: /system.slice/avahi-daemon.service
           ├─743 avahi-daemon: running [witherspoon.local
           └─780 avahi-daemon: chroot helpe

Jan 30 02:47:13 witherspoon avahi-daemon[743]: Registering new address record for 9.3.185.173 on eth0.IPv4.
Jan 30 02:47:14 witherspoon avahi-daemon[743]: Server startup complete. Host name is witherspoon.local. Local service cookie is 2200599522.
Jan 30 02:47:15 witherspoon avahi-daemon[743]: Service "witherspoon" (/services/ssh.service) successfully established.
Jan 30 02:47:15 witherspoon avahi-daemon[743]: Service "sftp" (/services/sftp.service) successfully established.
Jan 30 02:47:15 witherspoon avahi-daemon[743]: Service "witherspoon" (/services/sftp-ssh.service) successfully established.
Jan 30 02:47:15 witherspoon avahi-daemon[743]: Service "phosphor_rest" (/services/phosphor_rest.service) successfully established.
Jan 30 02:47:15 witherspoon avahi-daemon[743]: Service "obmc_console" (/services/obmc_console.service) successfully established.
Jan 30 02:47:34 witherspoon avahi-daemon[780]: ../../avahi-0.6.32/avahi-daemon/chroot.c: open() failed: No such file or directory
Jan 30 02:47:34 witherspoon avahi-daemon[743]: Failed to open /usr/share/dbus-1/interfaces/org.freedesktop.Avahi.Server.xml: Invalid argument
Jan 30 02:47:34 witherspoon avahi-daemon[743]: Failed to load introspection data.

ratagupt, 2017-03-17:
I updated my finding on https://github.com/openbmc/openbmc/issues/1251.
Both issues are same.

gkeishin, 2017-05-29:
Closing this as this is going get address with #1251  and I have confirmed the same with @ratagupt

closed 2017-05-29 by gkeishin