openbmc/openbmc #986: Setting sdbusplus properties causes core dump

anoo1, 2017-01-18:
With this sdbusplus change, setting dbus properties causes a core dump: https://github.com/openbmc/sdbusplus/commit/560e5fd5d135c7a3d4e354fd5346220fb9483575

1. Updated the phosphor-logging generated code with the sdbusplus tool that had the above change. The generate code was updated with these changes:
```
-        auto m = message::message(sd_bus_message_ref(value));
+        auto m = message::message(value);
```

2. Recompiled the phosphor-logging application and ran it on the BMC. The application now core dumps here when trying to set more than one object properties:
```
this->id(properties.id);  <= ok
this->message(properties.msg);  <= core dump
```

Core dump bt:
```
$ $GDB phosphor-log-managercore_phosphor-log-ma.1393
...
Core was generated by `/usr/sbin/phosphor-log-manager'.
Program terminated with signal SIGABRT, Aborted.
#0  0x4731d5d8 in two_way_long_needle () from /lib/libc.so.6
```

williamspatrick, 2017-01-19:
Is there a backtrace?  Is this merged code on the phosphor-logging side?

williamspatrick, 2017-01-19:
Oh. Did you regenerate the bindings?  I suspect you did not. The have a change in that same commit.

anoo1, 2017-01-19:
Talking with Patrick I forgot to update my sdk with the sdbusplus change. It works now, closing.

closed 2017-01-19 by anoo1