openbmc/openbmc #988: [Witherspoon - v1.99.0-400-g223aa02] Driver 453: Hit with frequent error dbus exception during GET enumerate request (9 out of 10 times fails)

sivassrr, 2017-01-19:
Using Driver 453 - Frequently hitting with dbus exception during GET enumerate request.  
Out of 10 iterations, 9 time failed.  Failure details as follows:

```
11:35:20.143 	INFO 	Get Request using : alias=openbmc, uri=/org/openbmc/enumerate, headers=None 	
11:35:20.144 	INFO 	${ret} = <Response [500]> 	
00:00:00.013 KEYWORD BuiltIn . Run Keyword If '${quiet}' == '${0}', Log Response, ${ret}
Documentation: 	

Runs the given keyword with the given arguments, if condition is true.
Start / End / Elapsed: 	20170119 11:35:20.145 / 20170119 11:35:20.158 / 00:00:00.013
00:00:00.005 KEYWORD rest_client . Log Response ${ret}
Start / End / Elapsed: 	20170119 11:35:20.153 / 20170119 11:35:20.158 / 00:00:00.005
00:00:00.001 KEYWORD ${msg} = BuiltIn . Catenate SEPARATOR=, Response code:, ${resp.status_code}, , Content:, ${resp.content}
Documentation: 	

Catenates the given items together and returns the resulted string.
Start / End / Elapsed: 	20170119 11:35:20.154 / 20170119 11:35:20.155 / 00:00:00.001
11:35:20.155 	INFO 	${msg} = Response code:500, Content:{
  "data": {
    "description": "Internal Server Error", 
    "exception": "DBusException(\"Unknown interface 'org.openbmc.Object.Delete'.\",)", 
    "traceback": [
"... 	
00:00:00.001 KEYWORD rest_client . Logging ${msg}, console=True 

Start / End / Elapsed: 	20170119 11:35:20.156 / 20170119 11:35:20.157 / 00:00:00.001
00:00:00.000 KEYWORD BuiltIn . Log ${msg}, console=True
Documentation: 	

Logs the given message with the given level.
Start / End / Elapsed: 	20170119 11:35:20.157 / 20170119 11:35:20.157 / 00:00:00.000
11:35:20.157 	INFO 	Response code:500, Content:{
  "data": {
    "description": "Internal Server Error", 
    "exception": "DBusException(\"Unknown interface 'org.openbmc.Object.Delete'.\",)", 
    "traceback": [
      "Traceback (most recent call last):", 
      "  File \"/usr/lib/python2.7/site-packages/bottle.py\", line 862, in _handle", 
      "    return route.call(**args)", 
      "  File \"/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py\", line 522, in __call__", 
      "    return self.callback(*a, **kw)", 
      "  File \"/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py\", line 678, in wrap", 
      "    return self.to_jsonp(callback(*a, **kw))", 
      "  File \"/usr/lib/python2.7/site-packages/bottle.py\", line 1734, in wrapper", 
      "    rv = callback(*a, **ka)", 
      "  File \"/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py\", line 613, in wrap", 
      "    resp = {'data': callback(*a, **kw)}", 
      "  File \"/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py\", line 87, in __call__", 
      "    return getattr(self, 'do_' + request.method.lower())(**kw)", 
      "  File \"/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py\", line 183, in do_get", 
      "    mapper_data=request.route_data['map']).dataitems()}", 
      "  File \"/usr/lib/python2.7/site-packages/obmc/mapper/bindings.py\", line 187, in enumerate_subtree", 
      "    path, owner, interfaces, match))", 
      "  File \"/usr/lib/python2.7/site-packages/obmc/mapper/bindings.py\", line 110, in __get_properties_on_bus", 
      "    properties_iface, i))", 
      "  File \"/usr/lib/python2.7/site-packages/obmc/mapper/bindings.py\", line 96, in __get_properties_on_iface", 
      "    properties_iface.GetAll, iface)", 
      "  File \"/usr/lib/python2.7/site-packages/obmc/mapper/bindings.py\", line 82, in __try_properties_interface", 
      "    return f(*a)", 
      "  File \"/usr/lib/python2.7/site-packages/dbus/proxies.py\", line 145, in __call__", 
      "    **keywords)", 
      "  File \"/usr/lib/python2.7/site-packages/dbus/connection.py\", line 651, in call_blocking", 
      "    message, timeout)", 
      "DBusException: org.freedesktop.DBus.Error.UnknownInterface: Unknown interface 'org.openbmc.Object.Delete'."
    ]
  }, 
  "message": "500 Internal Server Error", 
  "status": "error"
} 	
```

sivassrr, 2017-01-19:
As we use enumerate call during the suite setup, Entire suite is failing because of that.

sivassrr, 2017-01-19:
Observed issue in one system and another system issue is not observed.

sivassrr, 2017-01-19:
[witherspoonw1Dr453.pdf](https://github.com/openbmc/openbmc/files/715965/witherspoonw1Dr453.pdf)

vishwabmc, 2017-01-19:
@sivassrr I presume you would need to wait until @cklokma puts the fix for one of those.

@geissonator right ?

sivassrr, 2017-01-19:
[journalctldbusexception.txt](https://github.com/openbmc/openbmc/files/716547/journalctldbusexception.txt)

sivassrr, 2017-01-19:
Sure @vishwabmc. May I know what fixes @cklokma fixing. Any details will help

sivassrr, 2017-01-23:
Recreated on Dr468 and here is the FFDC.
[enumerateissue988.zip](https://github.com/openbmc/openbmc/files/723846/enumerateissue988.zip)

sivassrr, 2017-05-29:
With the latest driver not seeing this issue. So as of now, Closing it.

closed 2017-05-29 by sivassrr