openbmc/openbmc #989: Witherspoon: MAC address reset after doing a code update followed by power cycle

yadlapati, 2017-01-19:
Seeing MAC address lost on w1.aus.stglabs.ibm.com after doing a code update and then doing a power cycle.
steps to recreate:
1. flash new obmc build
2. system comes up fine and MAC address is correct
3. power cycle and MAC address is lost

root@witherspoon:~# ifconfig 
eth0      Link encap:Ethernet  HWaddr 72:B2:3E:BA:B8:C3  
          inet addr:9.3.185.29  Bcast:9.3.185.127  Mask:255.255.255.128

MAC is supposed to be 70:E2:84:14:16:A2

vishwabmc, 2017-01-20:
@anoo1 I remember you saving the MAC address onto some space and then fetching it back on a boot ?. Is this in the same area ?

anoo1, 2017-02-22:
Yeah, witherspoon has an assigned space in the boxelder vpd to store the mac address. There's a work item to add the logic to read it and restore it same as it's currently done in barreleye.

anoo1, 2018-05-23:
Implemented with:
https://github.com/openbmc/openbmc/issues/525
https://github.com/openbmc/openbmc/issues/1315

closed 2018-05-23 by anoo1