openbmc/openbmc #992: obmc-console-server doesn't relay to local-tty unless socket is open

GHF, 2017-01-19:
When I run obmc-console-server with a local-tty relay to ttyS0, I get these errors if I don't *also* open obmc-console-client:

```
Jan 19 22:08:35 zaius sh[769]: obmc-console-server: Write error: Resource temporarily unavailable
```

I also don't see any output on ttyS0.

These are my command lines:

```
             ├─system-obmc\x2dconsole.slice
             │ └─obmc-console@ttyVUART0.service
             │   ├─769 /bin/sh -c realpath /dev/ttyVUART0 | xargs basename | xargs /usr/sbin/obmc-console-server --config /etc/obmc-console.conf
             │   ├─777 xargs /usr/sbin/obmc-console-server --config /etc/obmc-console.conf
             │   └─779 /usr/sbin/obmc-console-server --config /etc/obmc-console.conf ttyS1
```

and this is the config file:

```
root@zaius:~# cat /etc/obmc-console.conf
lpc-address = 0x3f8
sirq = 4
local-tty = ttyS0
```

I haven't debugged this much yet, but I do get UART output when obmc-console-client is also connected.

GHF, 2017-01-20:
Hi Jeremy, it looks like you've uploaded a bunch of commits related to multiple client handling and flow control. Does that mean this bug will be fixed once the console clients can receive data independently?

jk-ozlabs, 2017-02-07:
@GHF: the console clients could always receive data independently, those changes just addressed an issue once we'd hit a full buffer.

This sounds like the same issue at #1096, in that we're writing faster than `ttyS0` can output.

GHF, 2017-02-09:
Oops, I think the lack of serial output is not related to obmc-console-client being connected, but rather /dev/ttyS0 being set to the wrong baud rate. I didn't realize that we're booting with default 9600 baud. I need to double check.

jk-ozlabs, 2017-02-09:
Do we want a config option (say, `local-tty-baud`) so that `obmc-console-server` can set the baud rate on startup?

@GHF: also, I've proposed a few changes to gerrit, which should fix some of the tty bufferring.

GHF, 2017-02-09:
I had intended to set a default baud rate in the device tree, but a `local-tty-baud` option works better since it makes more sense for the application to set the speed.

GHF, 2017-02-09:
heh, looks like adding `current-speed = <115200>;` to `uart1` in the device tree has no effect (or rather the divisor that sets seems to be overwritten by some other tty initialization).

So it looks like we need `local-tty-baud` after all.

shenki, 2017-02-10:
As I understand it `current-speed` is used to say "the bootloader has set it to this speed, don't change it". I'm not sure how we set the speed when it's not a console.

From https://www.kernel.org/doc/Documentation/devicetree/bindings/serial/8250.txt:
```
- current-speed : the current active speed of the UART.
```

GHF, 2017-02-10:
That explanation for the `current-speed` property makes a lot of sense.

I couldn't find anything either about configuring the kernel to set UART speed.

GHF, 2017-04-17:
I'm going to add an config flag to the tty handler to optionally change the baud rate of the terminal that it opens.

closed 2017-04-24 by williamspatrick