openbmc/openbmc #994: [Barreleye "v1.99.0-421-g48b42e0] BMC stop responding during test

gkeishin, 2017-01-20:
BMC stops responding in the middle of the test.  I dont have data or appropriate logs to show what actually happened.. but on the serial i see this when it was hung. I had to AC cycle it to recover.

barreleye login: ftgmac100 1e660000.ethernet eth0: NCSI interface up
random: nonblocking pool is initialized
ftgmac100 1e660000.ethernet eth0: NCSI interface down
ftgmac100 1e660000.ethernet eth0: NCSI interface up
ftgmac100 1e660000.ethernet eth0: NCSI interface down
ftgmac100 1e660000.ethernet eth0: NCSI interface up
ftgmac100 1e660000.ethernet eth0: NCSI interface up
ftgmac100 1e660000.ethernet eth0: NCSI interface down
ftgmac100 1e660000.ethernet eth0: NCSI interface up

Looks like the network test seems to be the point when this started..

gkeishin, 2017-01-20:
Thinking if such case happens in future.. what are the data we would need to collect for debugging ?
so that we can enhance the test FFDC to capture those.. any thoughts ?

gkeishin, 2017-01-31:
Driver : v1.99.1-55-g51ca97e

A power on  request to boot the host .. where the BMC stops to respond for a brief moment and onward the flaky respond until powered down..

```Jan 30 02:06:08 barreleye timemanager[873]:  Host_time_str: [ Mon Jan 30 02:06:08 2017 GMT ] host_time usec: [ 1485741968650549 ] host_offset: [ 0 ]
Jan 30 02:06:08 barreleye ipmid[704]: sscanf Failed in extracting mac address.
Jan 30 02:06:08 barreleye ipmid[704]: getHostNetworkData failed for get_sys_boot_options.
Jan 30 02:06:08 barreleye ipmid[704]: ERROR:[0xFF] handling NetFn:[0x0], Cmd:[0x9]
Jan 30 02:06:09 barreleye ipmid[704]: sscanf Failed in extracting mac address.
Jan 30 02:06:09 barreleye ipmid[704]: getHostNetworkData failed for get_sys_boot_options.
Jan 30 02:06:09 barreleye ipmid[704]: ERROR:[0xFF] handling NetFn:[0x0], Cmd:[0x9]
Jan 30 02:06:09 barreleye ipmid[704]: sscanf Failed in extracting mac address.
Jan 30 02:06:09 barreleye ipmid[704]: getHostNetworkData failed for get_sys_boot_options.
Jan 30 02:06:09 barreleye ipmid[704]: ERROR:[0xFF] handling NetFn:[0x0], Cmd:[0x9]
ftgmac100 1e660000.ethernet eth0: NCSI interface down
Jan 30 02:06:09 barreleye ipmid[704]: sscanf Failed in extracting mac address.
Jan 30 02:06:09 barreleye ipmid[704]: getHostNetworkData failed for get_sys_boot_options.
Jan 30 02:06:09 barreleye ipmid[704]: ERROR:[0xFF] handling NetFn:[0x0], Cmd:[0x9]
Jan 30 02:06:09 barreleye ipmid[704]: No registered handlers for NetFn:[0x3A], Cmd:[0x1A] trying Wilcard implementation
Jan 30 02:06:09 barreleye ipmid[704]: No Registered handlers for NetFn:[0x3A],Cmd:[0xFF]
Jan 30 02:06:09 barreleye ipmid[704]: ERROR:[0xC1] handling NetFn:[0x3A], Cmd:[0x1A]
Jan 30 02:06:09 barreleye kernel: ftgmac100 1e660000.ethernet eth0: NCSI interface down
Jan 30 02:06:09 barreleye kernel[697]: ftgmac100 1e660000.ethernet eth0: NCSI interface down
ftgmac100 1e660000.ethernet eth0: NCSI interface up
Jan 30 02:06:13 barreleye kernel: ftgmac100 1e660000.ethernet eth0: NCSI interface up
Jan 30 02:06:13 barreleye kernel[697]: ftgmac100 1e660000.ethernet eth0: NCSI interface up
ftgmac100 1e660000.ethernet eth0: [ISR] = 0x5f: NO_RXBUF RPKT_LOST
ftgmac100 1e660000.ethernet eth0: [ISR] = 0x5f: NO_RXBUF RPKT_LOST
ftgmac100 1e660000.ethernet eth0: [ISR] = 0x5f: NO_RXBUF RPKT_LOST
ftgmac100 1e660000.ethernet eth0: [ISR] = 0x5f: NO_RXBUF RPKT_LOST
ftgmac100 1e660000.ethernet eth0: [ISR] = 0x5f: NO_RXBUF RPKT_LOST
ftgmac100 1e660000.ethernet eth0: [ISR] = 0x5f: NO_RXBUF RPKT_LOST
ftgmac100 1e660000.ethernet eth0: [ISR] = 0x5f: NO_RXBUF RPKT_LOST
ftgmac100 1e660000.ethernet eth0: [ISR] = 0x5f: NO_RXBUF RPKT_LOST
ftgmac100 1e660000.ethernet eth0: [ISR] = 0x5f: NO_RXBUF RPKT_LOST
net_ratelimit: 176 callbacks suppressed
ftgmac100 1e660000.ethernet eth0: [ISR] = 0x5f: NO_RXBUF RPKT_LOST
ftgmac100 1e660000.ethernet eth0: [ISR] = 0x5f: NO_RXBUF RPKT_LOST
ftgmac100 1e660000.ethernet eth0: [ISR] = 0x5f: NO_RXBUF RPKT_LOST
ftgmac100 1e660000.ethernet eth0: [ISR] = 0x5f: NO_RXBUF RPKT_LOST
ftgmac100 1e660000.ethernet eth0: [ISR] = 0x5f: NO_RXBUF RPKT_LOST
ftgmac100 1e660000.ethernet eth0: [ISR] = 0x5f: NO_RXBUF RPKT_LOST
ftgmac100 1e660000.ethernet eth0: [ISR] = 0x5f: NO_RXBUF RPKT_LOST
ftgmac100 1e660000.ethernet eth0: [ISR] = 0x5f: NO_RXBUF RPKT_LOST
ftgmac100 1e660000.ethernet eth0: [ISR] = 0x5f: NO_RXBUF RPKT_LOST
ftgmac100 1e660000.ethernet eth0: [ISR] = 0x57: NO_RXBUF

```

- [x] Journal log 
[journald_v1.99.1-55-g51ca97e_ftgmac100.txt](https://github.com/openbmc/openbmc/files/741820/journald_v1.99.1-55-g51ca97e_ftgmac100.txt)

geissonator, 2017-01-31:
Our old friend https://github.com/openbmc/openbmc/issues/184 it looks like?  The "Jan 30 02:06:09 barreleye ipmid[704]: sscanf Failed in extracting mac address." messages are concerning though.  Was the MAC set ok on the system?

gkeishin, 2017-01-31:
Yeah.. its coming back to haunt us again.. Will keep watch on it..

shenki, 2017-02-01:
I suspect the errors from the ftgmac100 driver are because this machine has the same MAC address as another machine on the network.

As @geissonator pointed out, we need to work out why the MAC was not being set. There is a printf that shows us the MAC address string but it's not shown in the log output. https://github.com/openbmc/phosphor-host-ipmid/blob/master/chassishandler.cpp#L324 Do we know why?

shenki, 2017-02-01:
@gkeishin having the output of `ifconfig -a` would help in FFDC

Does your network test set the same MAC address for all test? You need to ensure it is unique or you will continue hitting these errors

gkeishin, 2017-02-01:
@shenki  In the latest one I updated it was just power on test..  no network thing here in this case.. The MAC seems to be correct..  the config is still same.. If you want to take a look on AES (BarPvT).

shenki, 2017-02-01:
```
Jan 30 20:10:01 barreleye ipmid[715]:  host_config.macaddress.c_str()=[]
```

It's trying to set an empty string as the MAC address. Is this your test code, or is it missing some configuration? Can you reproduce from a clean image?

gkeishin, 2017-02-01:
@shenki Sure.. I'll try again..

gkeishin, 2017-02-02:
one of our folks hit on palmetto as well https://github.com/openbmc/openbmc/issues/1064

gwshan, 2017-02-03:
Actually, it's same to #1064

gkeishin, 2017-02-03:
@gwshan okie.. Thanks... so I can close this one up and we can track with #1064  perhaps ?

gwshan, 2017-02-03:
@gkeishin yes, it's ok to duplicate this one to #1064

gkeishin, 2017-02-03:
Closing as suggested... Thanks

closed 2017-02-03 by gkeishin