openbmc/openbmc #996: IPMI Add SEL fails

anoo1, 2017-01-20:
IPMI Add SEL command fails, so currently none of the host errors are logged to the BMC. Using BMC level v1.99.0-394. Journal output:

```
IPMI Incoming: Seq 0x05, NetFn 0x0a, CMD: 0x44 
0x000000: 00 00 02 00 00 00 00 20 00 04 12 35 6f 02 00 01 ....... ...5o...
IPMI Handling ADD-SEL for record 0x0001
Failed to get  busname: No such file or directory
Failed to get  busname: No such file or directory
Failed to call the method: send_esel_to_dbus Connection reset by peer
Failed to send esel to dbus
IPMI Response:
0x000000: 00 00 01                                        ...             
Failed to call the method: Transport endpoint is not connected
Dest: :1.50, Path: /org/openbmc/HostIpmi/1
Failed to process bus: Connection reset by peer
```

vishwabmc, 2017-01-23:
Indication that mapper is not running.. What is the ps output ?. Any journal data on why mapper may not be running ?

tomjoseph83, 2017-02-07:
v1.99.1-97 - I verified this with a later driver on a barreleye. The Add SEL command was successful and did it report any error.

```
root@barreleye:/tmp# ./ipmitool -I dbus raw 0x0a 0x44 0x00 0x00 0x02 0x00 0x00 0x00 0x00 0x20 0x00 0x04 0x12 0x35 0x6f 0x02 0x00 0x01
IPMI Incoming: Seq 0x01, NetFn 0x2c, CMD: 0x00 
0x000000: 00                                              .               
IPMI GROUP EXTENTIONS
IPMI Response:
0x000000: 00 00                                           ..              
IPMI Incoming: Seq 0x02, NetFn 0x2c, CMD: 0x00 
0x000000: 03                                              .               
IPMI GROUP EXTENTIONS
IPMI Response:
0x000000: 00 00                                           ..              
IPMI Incoming: Seq 0x03, NetFn 0x0a, CMD: 0x44 
0x000000: 00 00 02 00 00 00 00 20 00 04 12 35 6f 02 00 01 ....... ...5o...
IPMI Handling ADD-SEL for record 0x0001

IPMI Response:
0x000000: 00 00 01                                        ...             
 00 01

```

tomjoseph83, 2017-02-09:
@anoo1 I am seeing the event get added.
```
root@barreleye:/tmp# dbus-send --system --print-reply --dest=org.openbmc.records.events /org/openbmc/records/events/291 org.freedesktop.DBus.Properties.GetAll string:"org.openbmc.record"
method return time=1485940943.121950 sender=:1.2 -> destination=:1.93 serial=112 reply_serial=2
   array [
      dict entry(
         string "message"
         variant             string "A systemevent has experienced a error from a diagnostic test"
      )
      dict entry(
         string "severity"
         variant             string "error from a diagnostic test"
      )
      dict entry(
         string "reported_by"
         variant             string "Host"
      )
      dict entry(
         string "time"
         variant             string "2017:02:01 09:13:06"
      )
      dict entry(
         string "debug_data"
         variant             array of bytes [
               00 00 df 00 00 00 00 20 00 04 12 35 6f aa 00 00
            ]
      )
   ]
```

tomjoseph83, 2017-02-09:
```
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.2-1"
VERSION_ID="v1.99.2-1-g2d058c0-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.2-1"
BUILD_ID="v1.99.2" ```

tomjoseph83, 2017-02-10:
@geissonator Spoke to @anoo1  about the observation. She was okie to have this closed for now as the investigation is complete. We can reopen this if the issue resurfaces again.

closed 2017-02-10 by geissonator