openbmc/openbmc #998: Kernel Stack got corrupted on OpenPower system

wghoffa, 2017-01-20:
On the 3rd IPL attempt, the BMC became unresponsive on this one socket system:

Kernel panic - not syncing: stack-protector: Kernel stack is corrupted in: 802e6108
CPU: 0 PID: 1723 Comm: zaius_vcs.sh Tainted: G        W       4.7.10-f26558191540830589fe03932d05577957670b8d #1
Hardware name: ASpeed SoC
[<801072f8>] (unwind_backtrace) from [<80105558>] (show_stack+0x10/0x14)
[<80105558>] (show_stack) from [<8016d9a4>] (panic+0xb8/0x24c)
[<8016d9a4>] (panic) from [<80110df8>] (__stack_chk_fail+0x10/0x14)
[<80110df8>] (__stack_chk_fail) from [<802e6108>] (i2c_smbus_xfer+0x508/0x54c)
[<802e6108>] (i2c_smbus_xfer) from [<ffffffff>] (0xffffffff)
---[ end Kernel panic - not syncing: stack-protector: Kernel stack is corrupted in: 802e6108

It appears the kernel stack got corrupted causing a kernel panic.

spinler, 2017-01-25:
@amboar Can you help on this?

amboar, 2017-01-28:
I'll take a look on Monday. @legoater or Brendan Higgins might be able to provide better insight though.

legoater, 2017-01-28:
This fix looks like a good candidate : https://github.com/torvalds/linux/commit/30f939feaeee23e21391cfc7b484f012eb189c3c

williamspatrick, 2017-01-29:
The smbus_data union is just a byte array. Having it be uninitialized shouldn't cause a stack overflow. That commit says the initialization is to prevent kernel stack contents from escaping to userspace, but I don't think it will solve what we are running into.

legoater, 2017-01-29:
euh yes. you are right. It won't bring much to this issue and I should refrain my self from making suggestions at 8PM on saturdays :)

amboar, 2017-01-30:
@mdmillerii might've encountered this a while back - digging through the git histories I found openbmc/linux@e1079d1470b58d822815d170ff729d9480aed43b. I've pushed a branch with it applied to openbmc/linux master: amboar/linux@586d8321def71af458e8b657778b4a7f79fbb2aa. I've build-tested it for g4 and g5 but haven't boot tested it; it might be worth netbooting if someone gets the opportunity.

amboar, 2017-01-30:
I spent the afternoon analysing buffer management in [i2c_smbus_xfer_emulated](https://github.com/amboar/linux/blob/586d8321def71af458e8b657778b4a7f79fbb2aa/drivers/i2c/i2c-core.c#L2892). To the best of my knowledge, and despite the hairiness of the code, there don't appear to be any problems. I also dug around in [i2c-aspeed](https://github.com/amboar/linux/blob/586d8321def71af458e8b657778b4a7f79fbb2aa/drivers/i2c/busses/i2c-aspeed.c): The control flow and locking looks reasonable, so I can only hazard that we're getting an unexpected interrupt whilst `bus->msg` is not NULL from the previous read. I don't know if that's possible. Maybe we should NULL `bus->msg` before firing the completion?

amboar, 2017-01-30:
> Maybe we should NULL bus->msg before firing the completion?

I just pushed amboar/linux@73dc5b990e3de829502d2acd8b684731ce6a6fab which should (at least, more legitimately than the WARN_ONCE test) prevent the bug if the problem is spurious interrupts.

@shenki @legoater @mdmillerii thoughts?

legoater, 2017-01-30:
@amboar I think your analysis is correct and that a possible race with an interrupt could corrupt the stack. It would help to reproduce with :   dyndbg="file drivers/i2c/busses/i2c-aspeed.c  +p" on the kernel command line. Is that possible ?

legoater, 2017-01-30:
the dyndbg would be too verbose. we need to rely on the warnings @amboar added to confirm the issue. I gave the Andrew's kernel a try on a palmetto and on an ast2500 evb. It looks good but could not reproduce.

amboar, 2017-01-30:
@legoater so you were booting with and without my patches? It should be possible to drop the WARN_ONCE() patch with amboar/linux@73dc5b9 as it should never be hit. In theory.

legoater, 2017-01-30:
@amboar, yes I booted with and without. regarding https://github.com/amboar/linux/commit/73dc5b9, may be we should find a way to validate the theory and spot the issue first ? We would need a repro case. I will try under qemu.

mdmillerii, 2017-01-30:
The previous issue that drove https://github.com/openbmc/linux/commit/e1079d1470b58d822815d170ff729d9480aed43b was determined to be the irq from the stop message  occurring separately from the data transfer interrupt and both received a byte.

I was going on the assumption (in other contexts) that a stack canary smash is a deep stack write.  The name zaius_vcs.sh sounds like its issuing commands to change a voltage and likely sends i2c commands; seeing what that does or a simple strace may give a clue to the trigger if this is repeatable.

amboar, 2017-01-31:
> The name zaius_vcs.sh sounds like its issuing commands to change a voltage and likely sends i2c commands; seeing what that does or a simple strace may give a clue to the trigger if this is repeatable.

Yeah, this was my thought too. `zaius_vcs.sh` unbinds the kernel driver, whacks the bus via the chardev for a bit to reconfigure the UCD9000 device, then re-binds the driver. Maybe there are issues with that, but it would be in a lot of common code.

> I was going on the assumption (in other contexts) that a stack canary smash is a deep stack write. 

Yes, but we have to get deep into the stack with input from somewhere. If that input is off the bus then we're going to be taking interrupts to pull bytes off the wire. I was looking at buffer management in [i2c_smbus_xfer_emulated](https://github.com/amboar/linux/blob/586d8321def71af458e8b657778b4a7f79fbb2aa/drivers/i2c/i2c-core.c#L2892) to determine whether or not this might have to be unexpected data, i.e. if there were any potential issues with short buffers. It doesn't appear to be, so it may be that we're simply not using the buffers properly e.g. due to a race.

amboar, 2017-01-31:
I've been trying to reproduce the issue by hammering an i2c bus on palmetto, both through drivers and via the chardev interface (simultaneously), using a kernel built from the tip of openbmc/linux@f26558191540830589fe03932d05577957670b8d, but so far have failed. @wghoffa do you have a reliable way to reproduce the issue? Maybe I need access to a Zaius?

amboar, 2017-02-01:
I've also been trying to reproduce this on a Zaius machine amongst general reboot testing, but haven't succeeded.

williamspatrick, 2017-02-02:
I thought they were seeing this regularly just by running zaius_vcs.sh over and over.

legoater, 2017-02-02:
Just got a crash on a witherspoon, after many ipls, 
```
CPU: 0 PID: 2597 Comm: ucd_disable_vcs Not tainted 4.7.10-f26558191540830589fe03932d05577957670b8d #1
Hardware name: ASpeed SoC
[<801072f8>] (unwind_backtrace) from [<80105558>] (show_stack+0x10/0x14)
[<80105558>] (show_stack) from [<8016d9a4>] (panic+0xb8/0x24c)
[<8016d9a4>] (panic) from [<80110df8>] (__stack_chk_fail+0x10/0x14)
[<80110df8>] (__stack_chk_fail) from [<802e65c8>] (i2c_smbus_xfer+0x508/0x54c)
[<802e65c8>] (i2c_smbus_xfer) from [<ffffffff>] (0xffffffff)
---[ end Kernel panic - not syncing: stack-protector: Kernel stack is corrupted in: 802e65c8
```

gkeishin, 2017-03-07:
We just hit this on 

```
root@witherspoon:~# cat /etc/version
20170306191553
root@witherspoon:~# cat /etc/os-release| grep VERSION_ID
VERSION_ID="v1.99.3-42-ge8c604b"
root@witherspoon:~# uname -a
Linux witherspoon 4.7.10-2763f3eb775cf760b1c0f28d283da6df6c42f061 #1 Mon Mar 6 19:20:35 UTC 2017 armv6l GNU/Linux
root@witherspoon:~#
```
`On powering on`

```
Mar 07 05:31:38 witherspoon kernel: ucd9000 11-0064: Device ID UCD90160|2.3.4.0000|110603
Mar 07 05:31:38 witherspoon kernel[775]: ucd9000 11-0064: Device ID UCD90160|2.3.4.0000|110603
ucd9000 11-0064: Failed to read number of active pages
ucd9000: probe of 11-0064 failed with error -74
Kernel panic - not syncing: stack-protector: Kernel stack is corrupted in: 8032f8a0

CPU: 0 PID: 1429 Comm: ucd_disable_vcs Not tainted 4.7.10-2763f3eb775cf760b1c0f28d283da6df6c42f061 #1
Hardware name: ASpeed SoC
[<801072f8>] (unwind_backtrace) from [<80105558>] (show_stack+0x10/0x14)
[<80105558>] (show_stack) from [<8016d06c>] (panic+0xb8/0x248)
[<8016d06c>] (panic) from [<80110df4>] (__stack_chk_fail+0x10/0x14)
[<80110df4>] (__stack_chk_fail) from [<8032f8a0>] (i2c_smbus_xfer+0x508/0x54c)
[<8032f8a0>] (i2c_smbus_xfer) from [<ffffffff>] (0xffffffff)
Rebooting in 1 seconds..

U-Boot 2016.07 (Feb 22 2017 - 17:20:20 +0000)

DRAM:  496 MiB
```

amboar, 2017-03-07:
Okay, as it turns out this is relatively easy to trigger (despite my previous attempts). One or two runs of the line below has reliably reproduced the stack corruption. Multiple for loops are used just to monitor progress.

```sh
D=/sys/bus/i2c/drivers/ucd9000/; for i in `seq 0 9`; do echo $((i * 10)); for i in `seq 1 10`; do echo 0-0064 > ${D}/unbind; echo 0-0064 > ${D}/bind; done; done
```

shenki, 2017-03-07:
With some more debugging turned on it looks like the crash happens just as we probe the device:

```
[  178.320000] ucd9000 0-0064: Device ID UCD90160|2.3.4.0000|110603
[  178.690000] kobject: 'hwmon' (bd487300): kobject_add_internal: parent: '0-0064', set: '(null)'
[  178.700000] kobject: 'hwmon0' (bdf7a410): kobject_add_internal: parent: 'hwmon', set: 'devices'
[  178.720000] kobject: 'hwmon0' (bdf7a410): kobject_uevent_env
[  178.740000] kobject: 'hwmon0' (bdf7a410): fill_kobj_path: path = '/devices/platform/ahb/ahb:apb/1e78a000.i2c/i2c-0/i2c-0/0-0064/hwmon/hwmon0'
[  178.760000] kobject: 'hwmon0' (bdf7a410): kobject_uevent_env
[  178.760000] kobject: 'hwmon0' (bdf7a410): fill_kobj_path: path = '/devices/platform/ahb/ahb:apb/1e78a000.i2c/i2c-0/i2c-0/0-0064/hwmon/hwmon0'
[  178.840000] kobject: 'hwmon' (bd487300): kobject_cleanup, parent bdf73428
[  178.890000] kobject: 'hwmon' (bd487300): auto cleanup kobject_del
[  178.890000] kobject: 'hwmon' (bd487300): calling ktype release
[  178.890000] kobject: 'hwmon': free name
[  178.890000] kobject: 'hwmon0' (bdf7a410): kobject_cleanup, parent   (null)
[  178.890000] kobject: 'hwmon0' (bdf7a410): calling ktype release
[  178.890000] kobject: 'hwmon0': free name
[  178.890000] ucd9000 0-0064: Device ID UCD90160|2.3.4.0000|110603
[  179.050000] Kernel panic - not syncing: stack-protector: Kernel stack is corrupted in: 803805e4
[  179.050000] 
[  179.050000] CPU: 0 PID: 985 Comm: sh Not tainted 4.7.10-00294-g238565fe2bb9 #284
[  179.050000] Hardware name: ASpeed SoC
[  179.050000] [<80107b64>] (unwind_backtrace) from [<80105c38>] (show_stack+0x10/0x14)
[  179.050000] [<80105c38>] (show_stack) from [<8018ca28>] (panic+0xc4/0x260)
[  179.050000] [<8018ca28>] (panic) from [<801124e8>] (__stack_chk_fail+0x10/0x14)
[  179.050000] [<801124e8>] (__stack_chk_fail) from [<803805e4>] (i2c_smbus_xfer+0x538/0x560)
[  179.050000] [<803805e4>] (i2c_smbus_xfer) from [<ffffffff>] (0xffffffff)
[  179.050000] Rebooting in 1 seconds..
```

But then when I sprinkled more printk's in the probe function, it appeared to happen after the remove completed:

```
[  913.630000] kobject: 'hwmon0' (bc2db210): kobject_cleanup, parent   (null)
[  913.630000] kobject: 'hwmon0' (bc2db210): calling ktype release
[  913.630000] kobject: 'hwmon0': free name
[  913.870000] Kernel panic - not syncing: stack-protector: Kernel stack is corrupted in: 803805e4
[  913.870000] 
[  913.870000] CPU: 0 PID: 1668 Comm: sh Not tainted 4.7.10-00295-ga71cc0f6f26b #286
[  913.870000] Hardware name: ASpeed SoC
[  913.870000] [<80107b64>] (unwind_backtrace) from [<80105c38>] (show_stack+0x10/0x14)
[  913.870000] [<80105c38>] (show_stack) from [<8018ca28>] (panic+0xc4/0x260)
[  913.870000] [<8018ca28>] (panic) from [<801124e8>] (__stack_chk_fail+0x10/0x14)
[  913.870000] [<801124e8>] (__stack_chk_fail) from [<803805e4>] (i2c_smbus_xfer+0x538/0x560)
[  913.870000] [<803805e4>] (i2c_smbus_xfer) from [<ffffffff>] (0xffffffff)
```

My current suspicion is that an i2c transaction is completing after we've removed the driver. However, I think that the transfers are blocking so that shouldn't happen.

amboar, 2017-03-07:
Some investigation that suggests that the `DEVICE_ID` command (0xfd) to the UCD occasionally returns all 0xff, causing the i2c core to overrun the 34-byte data buffer with a 255-byte broken response. This patch should help by avoiding trashing the stack and also cause the probe to fail:

```diff
From 3edb8dc3d292107a26c9224e05b7629068d352dc Mon Sep 17 00:00:00 2001
From: Andrew Jeffery <andrew@aj.id.au>
Date: Wed, 8 Mar 2017 00:45:14 +1030
Subject: [PATCH] i2c: aspeed: Terrible work-around for ucd90xx probe crash

---
 drivers/i2c/busses/i2c-aspeed.c | 4 ++++
 1 file changed, 4 insertions(+)

diff --git a/drivers/i2c/busses/i2c-aspeed.c b/drivers/i2c/busses/i2c-aspeed.c
index d9aef5ca4736..509cf33fb889 100644
--- a/drivers/i2c/busses/i2c-aspeed.c
+++ b/drivers/i2c/busses/i2c-aspeed.c
@@ -511,6 +511,10 @@ static void ast_i2c_master_xfer_done(struct ast_i2c_bus *bus)
 					AST_I2CD_RX_BYTE_BUFFER) >> 8;
 
 			if (bus->query_len) {
+				/* Dirty hack, but we don't support PMBus */
+				if (WARN_ON(data > I2C_SMBUS_BLOCK_MAX))
+					data = I2C_SMBUS_BLOCK_MAX;
+
 				bus->msg->len += data;
 				bus->query_len = false;
 				dev_dbg(bus->dev, "got rx len: %d\n",
-- 
2.9.3

```

gkeishin, 2017-03-15:
```
- [x] SYSTEM     : W42 ( Witherspoon - 2 Sockets - 16 DIMMs )
- [x] BUILD_ID   : witherspoon-20170314151623.all.tar ( MFG Build ) 
- [x] VERSION_ID : ibm-v1.99.3-g35f91b9-r0
- [x] Kernel     : Linux witherspoon 4.7.10-ec62e5e4b9bb4f4e9df718fa7aa4d452fa906e1e #1 Tue Mar 14 10:31:58 CDT 2017 armv6l GNU/Linux
- [x] PNOR VERSION:
IBM-witherspoon-ibm-OP9_v1.15_1.3
        op-build-v1.15-15-gce29449
        buildroot-2017.02-4-gc28573e
        skiboot-28e96ff
        hostboot-86bae1c
        linux-4.10.1-p976d063
        petitboot-v1.4.1-p3ec3d6e
        witherspoon-xml-065aaa2
        occ-4141b5f
        hostboot-binaries-4b44090
        capp-ucode-1bb7503
```

gkeishin, 2017-03-15:
Sorry for inconvenience cause.. Please ignore the above comment.. that is from MFG BMC build , so it doesnt have this fix.. I ll verify on the latest master build and verify..  Will keep posted.. Thanks..

gkeishin, 2017-03-16:
I don't see this failure in latest build
```
- [x] BUILD_ID   : https://openpower.xyz/job/openbmc-build/630/
- [x] VERSION_ID : v1.99.3-93-gf0a463a
- [x] Kernel     : Linux witherspoon 4.7.10-c72ced5a19ceae19c7d792ae3e982bee5607dab4 #1 Wed Mar 15 22:22:06 UTC 2017 armv6l GNU/Linux
- [x] PNOR VERSION:
IBM-witherspoon-ibm-OP9_v1.15_1.3
        op-build-v1.15-15-gce29449
        buildroot-2017.02-4-gc28573e
        skiboot-28e96ff
        hostboot-86bae1c
        linux-4.10.1-p976d063
        petitboot-v1.4.1-p3ec3d6e
        witherspoon-xml-065aaa2
        occ-4141b5f
        hostboot-binaries-4b44090
        capp-ucode-1bb7503
```

amboar, 2017-07-07:
Brendan's I2C driver is part of the 4.10 kernel which properly resolves this issue. It can be closed.

closed 2017-08-23 by spinler