openbmc/openbmc #999: mboxd: 'Unknown lvalue' warnings for Wants and After directives

amboar, 2017-01-23:
Discovered in the logs by @spinler:

```
Jan 20 22:11:29 zaius systemd[1]: Reloading.
Jan 20 22:11:30 zaius systemd[1]: [/lib/systemd/system/mboxd.service:7] Unknown lvalue 'Wants' in section 'Service'
Jan 20 22:11:30 zaius systemd[1]: [/lib/systemd/system/mboxd.service:8] Unknown lvalue 'After' in section 'Service'
```

This could be caused by the directives appearing in the wrong section of the unit file.

vishwabmc, 2017-01-23:
Wants=op-wait-power-off@%i.service
After=op-wait-power-off@%i.service

Should those be mapper-wait instead ?.

spinler, 2017-01-23:
@vishwabmc What are you referring to?

spinler, 2017-01-23:
@amboar Are you planning on fixing this yourself?

amboar, 2017-01-23:
@spinler yes, I'll clean it up today

amboar, 2017-01-25:
Patches are in gerrit: https://gerrit.openbmc-project.xyz/#/q/topic:mboxd-service

closed 2017-01-27 by amboar