openbmc/openbmc #1020: Default compiling for Barreleye failed (pflash.bb, do_compile failed, error: common/ast-sf-ctrl.c: No such file or directory)

yadron, 2017-01-24:
Default compiling for Barreleye failed (pflash.bb, do_compile failed, error: common/ast-sf-ctrl.c: No such file or directory).

This workaround helped (just run this "ln" command manually and then again "bitbake obmc-phosphor-image"):

`ln -sf ../build/tmp/work/armv5e-openbmc-linux-gnueabi/pflash/5.4.0-r0/git/hw/ast-bmc/ast-sf-ctrl.c ../build/tmp/work/armv5e-openbmc-linux-gnueabi/pflash/5.4.0-r0/git/external/common/ast-sf-ctrl.c
`

-- this ln command should happen automatically here:

```
vi ../build/tmp/work/armv5e-openbmc-linux-gnueabi/pflash/5.4.0-r0/git/external/common/rules.mk

 60
 61 common/ast-sf-ctrl.c : ../../hw/ast-bmc/ast-sf-ctrl.c | common
 62         $(Q_LN)ln -sf ../../hw/ast-bmc/ast-sf-ctrl.c common/ast-sf-ctrl.c
```

More details in file (including build error text):
[issue_ast-sf-ctrl.txt](https://github.com/openbmc/openbmc/files/726427/issue_ast-sf-ctrl.txt)

If you know how to fix it or what is the reason, please fix or tell here.

Thank you.

gtmills, 2018-05-31:
@yadron Still a problem or can we close?

gtmills, 2018-07-06:
Closing. no update in a year+ and barreleye is no longer supported. Please reopen if still an issue. thanks!

closed 2018-07-06 by gtmills