openbmc/openbmc #1021: ipmi-fru-parser : map custom IPMI properties

dkodihal, 2017-01-24:
Custom IPMI properties and their mappings would have to defined, to create core inventory objects, for example. FYI @ratagupt and @geissonator.

dkodihal, 2017-01-25:
This should also handle core 'presence'.

spinler, 2017-01-25:
I thought https://github.com/openbmc/openbmc/issues/781 should handle core presence...  Core presence gets sent down as sensor sets, not FRU writes.

dkodihal, 2017-01-25:
Okay, do you know if host sends sensor information before IPMI FRU? Thanks for pointing this out, will discuss about this further and post updates here.

dkodihal, 2017-01-26:
Map need to map processor EC level (a custom property).

ratagupt, 2017-01-30:
Are we interested in EC property?if not then we can close this issue.

ratagupt, 2017-01-30:
is CCIN equivalent to EC property?

geissonator, 2017-01-30:
We can drop ECID, but we need EC for now and put in processor revision.

ratagupt, 2017-02-07:
No,We do need to make the change in the code as custom property have differnrt format.

closed 2017-02-21 by geissonator