openbmc/openbmc #1026: Zaius: No such file or directory: .../tmp/sysroots/zaius/pkgdata/runtime-reverse/obmc-op-control-host

amboar, 2017-01-25:
```
ERROR: obmc-phosphor-image-1.0-r0 do_rootfs: Error executing a python function in exec_python_func() autogenerated:

The stack trace of python calls that resulted in this exception/failure was:
File: 'exec_python_func() autogenerated', lineno: 2, function: <module>
     0001:
 *** 0002:license_create_manifest(d)
     0003:
File: '/home/andrew/src/openbmc/openbmc/meta/classes/license.bbclass', lineno: 48, function: license_create_manifest
     0044:    pkg_dic = {}
     0045:    for pkg in sorted(image_list_installed_packages(d)):
     0046:        pkg_info = os.path.join(d.getVar('PKGDATA_DIR', True),
     0047:                                'runtime-reverse', pkg)
 *** 0048:        pkg_name = os.path.basename(os.readlink(pkg_info))
     0049:
     0050:        pkg_dic[pkg_name] = oe.packagedata.read_pkgdatafile(pkg_info)
     0051:        if not "LICENSE" in pkg_dic[pkg_name].keys():
     0052:            pkg_lic_name = "LICENSE_" + pkg_name
Exception: OSError: [Errno 2] No such file or directory: '/home/andrew/src/openbmc/openbmc/build/tmp/sysroots/zaius/pkgdata/runtime-reverse/obmc-op-control-host'

ERROR: obmc-phosphor-image-1.0-r0 do_rootfs: Function failed: license_create_manifest
ERROR: Logfile of failure stored in: /home/andrew/src/openbmc/openbmc/build/tmp/work/zaius-openbmc-linux-gnueabi/obmc-phosphor-image/1.0-r0/temp/log.do_rootfs.30973
ERROR: Task 10 (/home/andrew/src/openbmc/openbmc/meta-phosphor/common/recipes-phosphor/images/obmc-phosphor-image.bb, do_rootfs) failed with exit code '1'
```

amboar, 2017-01-25:
Here's the log:
[log.do_rootfs.30973.txt](https://github.com/openbmc/openbmc/files/728341/log.do_rootfs.30973.txt)

An interesting snippet is:

```

NOTE: Not found virtual-obmc-host-ctl by name, searching provides ...
NOTE: cmd: /home/andrew/src/openbmc/openbmc/build/tmp/sysroots/x86_64-linux/usr/bin/smart --log-level=warning --data-dir=/home/andrew/src/openbmc/openbmc/build/tmp/work/zaius-openbmc-linux-gnueabi/obmc-phosphor-image/1.0-r0/rootfs/var/lib/smart query --provides virtual-obmc-host-ctl --show-format='$name-$version' | sed -ne 's/ *Provides://p'
NOTE: Found providers for virtual-obmc-host-ctl: obmc-op-control-host-1.0-r1@armv5e
p9-host-start-1.0-r1@armv6
```

amboar, 2017-01-25:
This is strange, as `bitbake -e obmc-phosphor-image` gives:

```
...
#
# $PREFERRED_PROVIDER_virtual/obmc-host-ctl [2 operations]
#   set? /home/andrew/src/openbmc/openbmc/meta-openbmc-machines/meta-openpower/conf/machine/include/openpower.inc:21
#     "obmc-op-control-host"
#   set /home/andrew/src/openbmc/openbmc/meta-openbmc-machines/meta-openpower/conf/machine/include/p9.inc:2
#     "p9-host-start"
# pre-expansion value:
#   "p9-host-start"
PREFERRED_PROVIDER_virtual/obmc-host-ctl="p9-host-start"
...
```

amboar, 2017-01-25:
For the moment I've hacked around it with `ln -s ../runtime/p9-host-start obmc-op-control-host` in the relevant directory.

eddiejames, 2017-01-31:
Any update on this? I get this building for witherspoon too.

amboar, 2017-02-01:
@eddiejames I haven't done anything about it. Maybe @bradbishop, @williamspatrick or @geissonator can give us a hand?

amboar, 2017-02-06:
@eddiejames I found this

> This was due to switching between systemd in DISTRO_FEATURES and not. Cleaning systemd and udev and rebuilding resolved the issue.
> 
> Lesson: use different tmp when changing DISTRO_FEATURES.

http://lists.openembedded.org/pipermail/openembedded-core/2014-September/097458.html

closed 2018-05-23 by amboar