openbmc/openbmc #1027: Large copies from flash make BMC unresponsive

amboar, 2017-01-25:
For instance, mboxd copies the 64MB pnor image into RAM on startup and on SIGHUP, which leads to approximately 20s of unresponsiveness.

Mikey has a patch to work around the problem in mboxd but the flash driver is ultimately at fault.

legoater, 2017-01-25:
If the copy is splitted in chunks of a reasonable size,  say 16KB, the BMC should behave correctly.
 
I suppose the issue comes from the mtd layer which passes the read request directly to the spi-nor 
driver which then loops on I/O until it is completed. Not sure how we can fix this.

mdmillerii, 2017-01-26:
The mtd layer has a synchronous design point; so all operations must complete at return (except for erase).  This leave spi-nor to poll the spi controller driver for completion.  It is done with a cond_resched in the loop so another task could wake up but no access to the mtd file systems would be allowed.  (Agreeing with difficult to fix).

amboar, 2018-04-18:
This is less of an issue with mbox reading in smaller chunks of the flash at a time. Given the complexities in the MTD layer, I don't expect we'll ever address this in a meaningful way as there isn't much of a motivation.

closed 2018-04-18 by amboar