openbmc/openbmc #1028: phosphor-logging : have the error log server process callouts

dkodihal, 2017-01-25:
The error-log server should be able to identify callout metadata, and based on the metadata, form associations between the error event and one or more inventory objects to be called out. It would need to somehow map callout metadata to an inventory object, for eg IIC device path and address to a FRU owning that IIC bus.

FYI @geissonator.

dkodihal, 2017-01-26:
- [ ] Write up design proposal and discuss the same in openbmc community.
- [ ] Design : Error log server should be able to map callout metadata journal to known callouts.
- [ ] Design : From the callout metadata, errorlog server should retrieve inventory object(s) to be called out. This would be via the help of MRW scripts and system specific configuration.
- [ ] Design : Add callouts as association objects to error event.

geissonator, 2017-02-15:
https://gerrit.openbmc-project.xyz/#/c/2402/

closed 2017-02-15 by geissonator