openbmc/openbmc #1030: [Palmetto VERSION_ID= #475-b3ff4009f88634] BMC failed to ping after codeupdate due to Ramdisk image is corrupt or invalid

sivassrr, 2017-01-25:
@SrideviRamesh commented on [Wed Jan 25 2017](https://github.com/openbmc/openbmc-test-automation/issues/298)

Issue:

After BMC codeupdate 'Verify Ping and REST Authentication' failed after retrying for 30 minutes

BMC failed to ping even after trying PDU cycle power off/on

Log info:

Logs captured from serial port console

```
RAM:  496 MiB
WARNING: Caches not enabled
Flash: 32 MiB
In:    serial
Out:   serial
Err:   serial
Net:   aspeednic#0
Hit any key to stop autoboot:  0
## Loading kernel from FIT Image at 20080000 ...
   Using 'conf@1' configuration
   Trying 'kernel@1' kernel subimage
     Description:  Linux kernel
     Type:         Kernel Image
     Compression:  uncompressed
     Data Start:   0x20080124
     Data Size:    1518552 Bytes = 1.4 MiB
     Architecture: ARM
     OS:           Linux
     Load Address: 0x40008000
     Entry Point:  0x40008000
     Hash algo:    sha1
     Hash value:   679d0aa0ed258d070d166c2dabdb92c6cbcfbed0
   Verifying Hash Integrity ... sha1+ OK
Wrong Ramdisk Image Format
Ramdisk image is corrupt or invalid
ast#
ast#
ast#
ast# printenv
baudrate=115200
bootargs=console=ttyS4,115200n8 root=/dev/ram rw
bootcmd=bootm 20080000 20300000
bootdelay=2
ethact=aspeednic#0
ethaddr=A0:42:3F:25:17:2C
spi_dma=yes
stderr=serial
stdin=serial
stdout=serial
verify=yes

Environment size: 241/131068 bytes
ast#

```
[serial_port_logs.txt](https://github.com/openbmc/openbmc-test-automation/files/728993/serial_port_logs.txt)
Attached serial port console logs

sivassrr, 2017-01-25:
@SrideviRamesh Moved to right repository (openbmc).

sivassrr, 2017-01-25:
Even hit it on witherspoon with the same driver

sivassrr, 2017-01-25:
ast# version

U-Boot 2016.07 (Jan 18 2017 - 00:15:39 +0000)
arm-openbmc-linux-gnueabi-gcc (GCC) 5.3.0
GNU ld (GNU Binutils) 2.26.0.20160214
ast# base
Base Address: 0x00000000
ast# showvar
HUSH_VERSION=0.01

I was about to collect more from witherspoon system, but by that time some one rebooted it using PDU.

geissonator, 2017-01-25:
@kc8apf I know some changes from you went in here recently? All 3 of our systems that tried the new build today hit this. I have seen an issue where our code update method doesn't seem to update the bootloader, maybe that's the issue here? #526 is that issue.

kc8apf, 2017-01-25:
@geissonator it looks like U-Boot is reading the FIT OK so u-boot should be fine.  What commit was this build built at?

geissonator, 2017-01-25:
b3ff400 on https://github.com/openbmc/openbmc/commits/master

geissonator, 2017-01-25:
Angel got the latest code manually programmed on the system (2a0cb54) and we boot up to BMC_READY fine so definitely related to #526

geissonator, 2017-01-31:
We're still struggling on this one.  It's causing a huge headache in the lab and bricking systems on a pretty regular basis.

Today I tried something new, I was on an image that did not have the new FIT support.  I updated u-boot on the system to something that did have the FIT support.  All was good.  I then applied the witherspoon.all.tar (latest master code) to the system.  I still got this error:

```
U-Boot 2016.07 (Jan 30 2017 - 13:21:35 -0600)

DRAM:  496 MiB
Flash: 32 MiB
In:    serial
Out:   serial
Err:   serial
Net:   aspeednic#0
Error: aspeednic#0 address not set.

Hit any key to stop autoboot:  0 
## Loading kernel from FIT Image at 20080000 ...
   Using 'conf@1' configuration
   Trying 'kernel@1' kernel subimage
     Description:  Linux kernel
     Type:         Kernel Image
     Compression:  uncompressed
     Data Start:   0x20080128
     Data Size:    1536920 Bytes = 1.5 MiB
     Architecture: ARM
     OS:           Linux
     Load Address: 0x80001000
     Entry Point:  0x80001000
     Hash algo:    sha1
     Hash value:   515dc645ff8c9630f8bc92f3cf19b48f45abc081
   Verifying Hash Integrity ... sha1+ OK
Wrong Ramdisk Image Format
Ramdisk image is corrupt or invalid
ast# 
```

Patrick's theory is that something didn't get updated in the all.tar packaging to point to the new FIT images?
@mdmillerii @kc8apf ?

shenki, 2017-02-05:
I downloaded `witherspoon-20170203124939.all.tar` which is `#510-2a46570ced7a9d` (2a46570ced7a9d) and it booted fine.

This suggests the update process is not working correctly for you.

```
qemu-system-arm -m 512 -M romulus-bmc -nodefaults -nographic -serial stdio \
	-net nic -net user,hostfwd=:127.0.0.1:2222-:22,hostname=qemu \
	-drive file=image-bmc,format=raw,if=mtd \
	-drive file=image-bmc,format=raw,if=mtd

U-Boot 2016.07 (Jan 31 2017 - 20:42:15 +0000)

DRAM:  496 MiB
Flash: 32 MiB
*** Warning - bad CRC, using default environment

In:    serial
Out:   serial
Err:   serial
Net:   aspeednic#0
Error: aspeednic#0 address not set.

Hit any key to stop autoboot:  0 
## Loading kernel from FIT Image at 20080000 ...
   Using 'conf@1' configuration
   Trying 'kernel@1' kernel subimage
     Description:  Linux kernel
     Type:         Kernel Image
     Compression:  uncompressed
     Data Start:   0x20080128
     Data Size:    1538528 Bytes = 1.5 MiB
     Architecture: ARM
     OS:           Linux
     Load Address: 0x80001000
     Entry Point:  0x80001000
     Hash algo:    sha1
     Hash value:   0892721b89470b93380bd168f01af7a909defcc2
   Verifying Hash Integrity ... sha1+ OK
## Loading ramdisk from FIT Image at 20080000 ...
   Using 'conf@1' configuration
   Trying 'ramdisk@1' ramdisk subimage
     Description:  obmc-phosphor-initramfs
     Type:         RAMDisk Image
     Compression:  lzma compressed
     Data Start:   0x201fd150
     Data Size:    1569194 Bytes = 1.5 MiB
     Architecture: ARM
     OS:           Linux
     Load Address: unavailable
     Entry Point:  unavailable
     Hash algo:    sha1
     Hash value:   5bbf1a7118eb1548e3f84374d15fb840ff058033
   Verifying Hash Integrity ... sha1+ OK
## Loading fdt from FIT Image at 20080000 ...
   Using 'conf@1' configuration
   Trying 'fdt@1' fdt subimage
     Description:  Flattened Device Tree blob
     Type:         Flat Device Tree
     Compression:  uncompressed
     Data Start:   0x201f7bfc
     Data Size:    21650 Bytes = 21.1 KiB
     Architecture: ARM
     Hash algo:    sha1
     Hash value:   600f8696be93987c56c259aa22828fd736f3261e
   Verifying Hash Integrity ... sha1+ OK
   Booting using the fdt blob at 0x201f7bfc
   Loading Kernel Image ... OK
   Loading Ramdisk to 9ea09000, end 9eb881aa ... OK
   Loading Device Tree to 9ea00000, end 9ea08491 ... OK

Starting kernel ...

Uncompressing Linux... done, booting the kernel.
Booting Linux on physical CPU 0x0
Linux version 4.7.10-f26558191540830589fe03932d05577957670b8d (jenkins@hudson) (gcc version 5.3.0 (GCC) ) #1 Fri Feb 3 12:52:20 UTC 2017
CPU: ARMv6-compatible processor [410fb767] revision 7 (ARMv7), cr=00c5387d
```

shenki, 2017-02-05:
I tested palmetto too for completeness:

```
$ qemu-system-arm -m 256 -M palmetto-bmc \
	-nodefaults -nographic -serial stdio \
	-net nic -net user,hostfwd=:127.0.0.1:2222-:22,hostname=qemu \
	-drive file=image-bmc,format=raw,if=mtd \
	-drive file=palmetto.pnor,format=raw,if=mtd

DRAM Init-DDR3
CBR0-1357135713571357
CBR134Done

U-Boot 2016.07 (Jan 31 2017 - 20:41:10 +0000)

DRAM:  240 MiB
WARNING: Caches not enabled
Flash: 32 MiB
*** Warning - bad CRC, using default environment

In:    serial
Out:   serial
Err:   serial
Net:   aspeednic#0
Error: aspeednic#0 address not set.

Hit any key to stop autoboot:  0 
## Loading kernel from FIT Image at 20080000 ...
   Using 'conf@1' configuration
   Trying 'kernel@1' kernel subimage
     Description:  Linux kernel
     Type:         Kernel Image
     Compression:  uncompressed
     Data Start:   0x20080124
     Data Size:    1518512 Bytes = 1.4 MiB
     Architecture: ARM
     OS:           Linux
     Load Address: 0x40008000
     Entry Point:  0x40008000
     Hash algo:    sha1
     Hash value:   076ed3c55dba277d4d9eacbf67e70f8aedbf9753
   Verifying Hash Integrity ... sha1+ OK
## Loading ramdisk from FIT Image at 20080000 ...
   Using 'conf@1' configuration
   Trying 'ramdisk@1' ramdisk subimage
     Description:  obmc-phosphor-initramfs
     Type:         RAMDisk Image
     Compression:  lzma compressed
     Data Start:   0x201f7fa0
     Data Size:    1567353 Bytes = 1.5 MiB
     Architecture: ARM
     OS:           Linux
     Load Address: unavailable
     Entry Point:  unavailable
     Hash algo:    sha1
     Hash value:   98ecf67457309f8953c1f7762b85cd64e6cc5e4c
   Verifying Hash Integrity ... sha1+ OK
## Loading fdt from FIT Image at 20080000 ...
   Using 'conf@1' configuration
   Trying 'fdt@1' fdt subimage
     Description:  Flattened Device Tree blob
     Type:         Flat Device Tree
     Compression:  uncompressed
     Data Start:   0x201f2dc8
     Data Size:    20760 Bytes = 20.3 KiB
     Architecture: ARM
     Hash algo:    sha1
     Hash value:   40d874797acc7f7b35238682ca2f6086895dd638
   Verifying Hash Integrity ... sha1+ OK
   Booting using the fdt blob at 0x201f2dc8
   Loading Kernel Image ... OK
   Loading Ramdisk to 4ea09000, end 4eb87a79 ... OK
   Loading Device Tree to 4ea00000, end 4ea08117 ... OK

Starting kernel ...

Uncompressing Linux... done, booting the kernel.
Booting Linux on physical CPU 0x0
Linux version 4.7.10-f26558191540830589fe03932d05577957670b8d (jenkins@hudson) (gcc version 5.3.0 (GCC) ) #1 Fri Feb 3 12:43:20 UTC 2017
```

geissonator, 2017-02-06:
The issue is when you have a level of code on your system that does not have the latest FIT image and you then follow the directions to update the all.tar via REST up here - https://github.com/openbmc/docs/blob/master/code-update.md

This wont recreate in QEMU because you're simply booting the new image, without going through the update process.

shenki, 2017-02-06:
I am aware that you're seeing an issue after updating. I demonstrated that the image is fine, and loads correctly. So the bug must be in the update process.

mdmillerii, 2017-02-06:
If one netboots the latest fit with the kernel / dtb / initrd combined then it will get the latest partitioning information in the dtb, and the image should flash (unless parts are missing from all.tar).

This is needed any time the flash partitioning changes.

The alternative is to copy a tarball with a single -bmc image and write the whole flash.

The rules will change with UBI.

sivassrr, 2017-09-19:
With latest Palmetto changes, we are not seeing this issue. So closing it. 
Thank you @swe12345 for your information. Will open issue if you see it.

closed 2017-09-19 by sivassrr