openbmc/openbmc #1031: [ Barreleye ] - Hitting busy processing bus traffic error while checking system state just after setting time owner

rahulmah, 2017-01-25:
Continuously hitting "busy processing bus traffic" error (503 Service Unavailable) while checking system state just after setting time owner from BMC to HOST

1. Make sure Time owner is set to BMC and time mode is set to NTP

2. Set time owner from BMC to HOST
```
07:09:42 Set Host Time With Host And Manual :: Set host time when time owne... URI:https://9.3.23.32/org/openbmc/settings/host0/attr/time_owner, method:Put, args:{u'data': {u'data': u'HOST'}, u'headers': {u'Content-Type': u'application/json'}}
07:09:43 Response code:200, Content:{
07:09:43   "data": null, 
07:09:43   "message": "200 OK", 
07:09:43   "status": "ok"
07:09:43 }
```
3. Now try to check system state. (here will see that busy processing bus traffic error)
```
07:09:44 URI:https://9.3.23.32/org/openbmc/managers/System//action/getSystemState, method:Post, args:{u'data': {u'data': []}, u'headers': {u'Content-Type': u'application/json'}}
07:09:44 Response code:503, Content:{
07:09:44   "data": {
07:09:44     "description": "org.freedesktop.DBus.Error.ObjectPathInUse: busy processing bus traffic"
07:09:44   }, 
07:09:44   "message": "503 Service Unavailable", 
07:09:44   "status": "error"
07:09:44 }
```

FFDC:
[Journal log for busy bus error.txt](https://github.com/openbmc/openbmc/files/729480/Journal.log.for.busy.bus.error.txt)
[20170123070944774429_SetHostTimeWithHostAndManual.zip](https://github.com/openbmc/openbmc/files/729483/20170123070944774429_SetHostTimeWithHostAndManual.zip)

rahulmah, 2017-01-25:
Same issue is also seen with Palmetto with latest build v1.99.1-7-gc897aa9-dirty

sivassrr, 2017-01-25:
Is it not same as https://github.com/openbmc/openbmc/issues/973

williamspatrick, 2017-03-10:
Already fixed.

closed 2017-03-10 by williamspatrick