openbmc/openbmc #1034: [Witherspoon - v1.99.1-2-g047ea60] Time mode is not getting set (inconsistent)

rahulmah, 2017-01-25:
Time mode is not getting set sometimes. See below step to recreate for more info.

Steps to recreate.
Note : This is one of the case where this issue is seen inconsistently
1. Set BMC owner from BMC to SPLIT
```
URI:https://xx.xx.xx.xx:443/org/openbmc/settings/host0/attr/time_owner, method:Put, args:{u'data': {u'data': u'SPLIT'}, u'headers': {u'Content-Type': u'application/json'}}
Response code:200, Content:{
  "data": null, 
  "message": "200 OK", 
  "status": "ok"
}
```

2. Power off the system for the setting to get applied
```
URI:https://xx.xx.xx.xx:443/org/openbmc/managers/System//action/getSystemState, method:Post, args:{u'data': {u'data': []}, u'headers': {u'Content-Type': u'application/json'}}
Response code:200, Content:{
  "data": "HOST_POWERED_OFF", 
  "message": "200 OK", 
  "status": "ok"
}
```

3. Check if time owner is set correct in both timemanager and host settings (here you will see that it is set correctly)
```
URI:https://xx.xx.xx.xx:443/org/openbmc/settings/host0/attr/time_owner, method:Get, args:{}
Response code:200, Content:{
  "data": "SPLIT", 
  "message": "200 OK", 
  "status": "ok"
}

URI:https://xx.xx.xx.xx:443/org/openbmc/TimeManager/attr/curr_time_owner, method:Get, args:{}
Response code:200, Content:{
  "data": "SPLIT", 
  "message": "200 OK", 
  "status": "ok"
}
```

4. Now set time mode to NTP. (here you will see that time mode is set correctly in host setting but now in time manage)
```
URI:https://xx.xx.xx.xx:443/org/openbmc/settings/host0/attr/time_mode, method:Put, args:{u'data': {u'data': u'NTP'}, u'headers': {u'Content-Type': u'application/json'}}
Response code:200, Content:{
  "data": null, 
  "message": "200 OK", 
  "status": "ok"
}
URI:https://xx.xx.xx.xx:443/org/openbmc/settings/host0/attr/time_mode, method:Get, args:{}
Response code:200, Content:{
  "data": "NTP", 
  "message": "200 OK", 
  "status": "ok"
}
URI:https://xx.xx.xx.xx:443/org/openbmc/TimeManager/attr/curr_time_mode, method:Get, args:{}
Response code:200, Content:{
  "data": "MANUAL", 
  "message": "200 OK", 
  "status": "ok"
```

FFDC: 
[20170124191405318790_SetBMCTimeWithSplitAndNTP.zip](https://github.com/openbmc/openbmc/files/730081/20170124191405318790_SetBMCTimeWithSplitAndNTP.zip)

gtmills, 2018-05-23:
@mine260309 Can we close? This is the old org time interface and a very old driver v1.99.1-2-g047ea60.

rahulmah, 2018-05-24:
Yeah this is an old issue with org interface. With new xyz interface we did not hit the problem. So closing the issue.

closed 2018-05-24 by rahulmah