openbmc/openbmc #1037: Tool for PNOR-to-OpenBMC format

anoo1, 2017-01-25:


anoo1, 2017-01-25:
Identify the partitions in the PNOR image (TOC, HBB, etc).
Identify which are read-only and which are read-write.
Identify which read-write partition is 'sideless' if any.
Map the PNOR image to the BMC filesystem layout (CRAMFS).

williamspatrick, 2017-01-26:
Also need to calculate (or estimate) a size of the ubifs volume(s) to create to store the rw data.

williamspatrick, 2017-01-26:
Final output is a tool that generates a SquashFS image containing all the pieces of PNOR (as individual files) and some metadata descriptor file that allows us to recreate a TOC and allocate rw volumes.

gtmills, 2017-01-31:
- [ ]   Determine requirements of structures
- [ ]   Initial setup of file structure
- [ ]   Implement tool to create SquashFS image
- [ ]   Test code

anoo1, 2017-01-31:
Per Patrick, it'd be best if this tool is built around pflash.

gtmills, 2017-01-31:
Tool will be in C++ and will go in https://github.com/openbmc/openpower-pnor-code-mgmt.

shenki, 2017-02-01:
It should only be a one liner to pack the files into a squashfs image.

The process should be modifying the op-build build process to create a squashfs with the ro components and their manifest, along side the PNOR. This squashfs could then be consumed by the openbmc build process.

stewart-ibm, 2017-02-01:
op-build should generate what's needed. If you need to write a C++ to do that, something has gone terribly wrong. It should just be outputting a file or a filesystem image.

I would argue that any read/write partitions should be able to be initialized by the bits of code using them (so don't need to be created) - and each of those come with appropriate userspace tools to init/read/write so we can adequately unit test code that parses them.

the only exception is the secure boot keystore, which you need to be able to boot at all with secure boot.

I gather the use case here is for the mbox daemon to be able to read the host firmware and present it to the host in some format?

jk-ozlabs, 2017-02-01:
Also, why squashfs?

If it's because that's how the BMC will store the image in its own flash, then we're unnecessarily tying the BMC implementation into the firmware image format (and then it means that the on-flash storage will be read-only, no?)

If it's just a format picked to transfer a bunch of files to the BMC, there are better choices, with tools that are more suited to our use-case. I'd suggest a simple compressed CPIO archive.

jk-ozlabs, 2017-02-01:
... not that I'm objecting to using squashfs, but want to make sure we're making the right choices here - the firmware package format will be a fairly long-standing decision.

amboar, 2017-02-01:
> Also, why squashfs? If it's because that's how the BMC will store the image in its own flash, then we're unnecessarily tying the BMC implementation into the firmware image format (and then it means that the on-flash storage will be read-only, no?)

Only if the entire PNOR was a squashfs image. My understanding is that it will instead be partitioned into multiple UBI volumes, one might contain the suggested squashfs image and be RO, while other(s) might be write-capable. There are other issues that indicate this is the design: #1035 #1048 

> If it's just a format picked to transfer a bunch of files to the BMC, there are better choices, with tools that are more suited to our use-case. I'd suggest a simple compressed CPIO archive.

I agree, I'd lean towards something simpler for that case.

williamspatrick, 2017-02-02:
@shenki -
```
The process should be modifying the op-build build process to create a squashfs with the ro components and their manifest, along side the PNOR. This squashfs could then be consumed by the openbmc build process.
```

Why would we modify the op-build process for something that is currently openbmc specific?  A particular machine (ex. Romulus) can run both OpenBMC and not.  The output of op-build should continue to be a "what the host would like to see for a PNOR layout".

@stewart-ibm 
```
I would argue that any read/write partitions should be able to be initialized by the bits of code using them (so don't need to be created) - and each of those come with appropriate userspace tools to init/read/write so we can adequately unit test code that parses them.
```

I'm not entirely following your question.  The intention here is that the tool will generate a SquashFS representation of the "clean from factory" state of the FFS partiions.  Likely for each FFS partition "FOO" there will be a file on the SquashFS image named "FOO".

It is quite expensive to generate a SquashFS file system on the fly on the BMC.  It also, in the long term, makes signature checking on the BMC difficult since FIT, dm-verity, etc. would likely desire those signatures to have been generated by the build system.

@jk-ozlabs 
```
Also, why squashfs?

If it's because that's how the BMC will store the image in its own flash, then we're unnecessarily tying the BMC implementation into the firmware image format (and then it means that the on-flash storage will be read-only, no?)
```

I'm not really understanding your point either.  We are _not_ modifying op-build.  This is an openbmc issue and not an open-power issue.  The format is tied to the software being developed for OpenBMC for how the BMC will internally manage the PNOR.

SquashFS because that is the "standard" for read-only partitions in the Linux flash world right now.

```
... not that I'm objecting to using squashfs, but want to make sure we're making the right choices here - the firmware package format will be a fairly long-standing decision.
```

Again, I don't really understand why you're suggesting it is a long-standing decision.  In fact, if another vendor doesn't want SquashFS but wants to use JFFS2, it should be pretty trivial to change over to that.

@amboar 
```
Only if the entire PNOR was a squashfs image. My understanding is that it will instead be partitioned into multiple UBI volumes, one might contain the suggested squashfs image and be RO, while other(s) might be write-capable.
```

Yep, you got it.

```
Edit: I may have misinterpreted with that response, squashfs is still awkward for updating individual files.

If it's just a format picked to transfer a bunch of files to the BMC, there are better choices, with tools that are more suited to our use-case. I'd suggest a simple compressed CPIO archive.
I agree, I'd lean towards something simpler for that case.
```

I get the impression that you 4 are worried about being an open-power host firmware developer and trying to get your temporary testing image onto the system.  Adriana's email on the mailing list I believe talks about having an alternative location on the BMC to load partitions from.  For a read-write FFS partition, the search location order might be something like:
* /usr/local/pnor  (read-write temporary location)
* /mnt/pnor-rw (read-write permanent location)
* /mnt/pnor-ro (read-only original state location)
If the first mount point where we find the file "FOO" will be used to load the FFS partition "FOO" in the virtual PNOR presented to the Host.

I don't recall if we documented this thought process, but we also discussed having a compile-time option and/or runtime setting to disable some of these search paths.  For example, on a field deployed  system you might not want end-users to put temporary images into /usr/local even if they had access.

amboar, 2017-02-03:
> I get the impression that you 4 are worried about being an open-power host firmware developer and trying to get your temporary testing image onto the system. Adriana's email on the mailing list I believe talks about having an alternative location on the BMC to load partitions from. 

Right. I think the point is why not simply use a RW-capable filesystem for the whole PNOR? You could make specific partition files RO as necessary, and that way we wouldn't need the complexity of the location search order. Secure/trusted boot on the host would verify the content of the partitions anyway, so I don't think the fact that the PNOR is writeable by the BMC is an issue - it is anyway even with a SquashFS, just at a much coarser level.

williamspatrick, 2017-02-03:
I have two reasons for desiring a RO file system:

1. A common problem we experience in manufacturing is that they need to load patches on to workaround some late concern. Just before shipping the machine they want to run something to easily reset the machine to clean state. A RO file-system gives us an easy way to do that. (This also goes for clearing out RW data that might have been set during the MFG runs). 

2. SquashFS compresses significantly better than UBIFS or JFFS2. Both of those RW file systems have overhead in their structures as well due to needing to support writes. 

We can accomplish the patch system with an overlay instead of search order, which is effectively what we do today with the BMC.

williamspatrick, 2017-02-03:
> Secure/trusted boot on the host would verify the content of the partitions anyway

Some vendors have already mentioned also having the BMC do verification as part of their design. IBM probably won't do it on our machines but we should facilitate others adding that code if they desire and it is relatively easy for us to do so. Now, you could argue that they can use the same SB algorithms as the host but the one I've talked to in particular wants to use a different signature algorithm on BMC content.

jk-ozlabs, 2017-02-07:
Hi @williamspatrick - from your comment:

> I'm not really understanding your point either. We are not modifying op-build. This is an openbmc issue and not an open-power issue. The format is tied to the software being developed for OpenBMC for how the BMC will internally manage the PNOR.

OK, this was a bit I was missing. The consequence is that we're introducing an extra step in the build process -  it's not possible to just using the output of `op-build` to flash onto a machine. Instead, (as you say), there's a step required, using by a different tool, to prepare a firmware build into something flashable.

Are we okay with that?

If so, I have no issue with using a BMC-specific image format - and if we're using SquashFS, then that's what makes sense to use.

gtmills, 2017-03-01:
Last commit is https://gerrit.openbmc-project.xyz/#/c/2326/

closed 2017-03-02 by rfrandse