openbmc/openbmc #1038: OpenFSI: Provide user space interface for PDBG

cbostic, 2017-01-26:
PDBG will need to transition from controlling GPIO lines directly via sysfs to instead utilize the OpenFSI driver so that there isn't contention between the two GPIO users.   Plan is to create a character device for FSI that can be opened with read/write ops triggered by getcfam and putcfam PDBG commands.

shenki, 2017-01-30:
To clarify, pdbg does not use sysfs. It directly writes to the hardware using `/dev/mem`.

This userspace API must be reviewed by @apopple before any code is written against it.

cbostic, 2017-01-30:
PDBG putcfam( ) and getcfam( ) appear to be good locations to make the transition into the openFSI driver.  Inside of these @apopple could open a FSI driver character dev and then perform a read or write syscall as appropriate.  Hopefully we can discuss this further in tonight's sync up call.

cbostic, 2017-02-01:
As per last Monday evening's (Central U.S time) call the plan for PDBG on OpenFSI will be for the application to access the read/write operations via the sysfs debug interface @shenki added.   The idea of using a fsi char dev to open/read/write to may eventually be something to explore further.

shenki, 2017-02-06:
This is blocked by https://github.com/openbmc/openbmc/issues/1086

We need @cbostic to look at this issue and resolve the problem with the kernel driver before pdbg work can proceed.

rfrandse, 2017-02-28:
Done

closed 2017-02-28 by rfrandse